PX4/PX4-Autopilot #3626: Thrust jumps up after switching to posctrl

Thomas10111, 2016-01-28:
We were using the px4flow as position estimator. Optical flow is integrated over time offboard and then fed into the /mavros/vision_pose/pose node, we use filtered sonar data as z. We are only flying indoors. After switching tp position control the quad immediately increased its height, switching back to manual mode let to a drop of the craft and finally because it was tethered to a crash. See log file
http://logs.uaventure.com/view/qbNWoN3taic7rSHjTChDyC#Thrust_PLOT

ecmnet, 2016-01-29:
@Thomas10111 Did you have a look on https://github.com/PX4/Firmware/pull/3549?

tumbili, 2016-01-29:
@Thomas10111 To me it looks like you have a problem with estimation. Compare your z velocity with your z height, they don't make sense at all to me. Your velocity looks wrong, during the entire flight it has a constant offset.

mhkabir, 2016-01-29:
@Thomas10111 Are you rotating the velocities with the body yaw properly in your estimator? Because the velocities are reported in body frame from the Flow module

If you just want PX4Flow based position control, you can just try INAV or LPE. No need for a companion computer/ROS.

Thomas10111, 2016-01-29:
Thanks for your replies. Estimation could indeed be a problem, I use the flow from /px4flow/opt_flow integrate it over time, average the z position (20 values) and publish the position to /mavros/vision_pose/pose. So, I am not correcting for the yaw. I expected to get some deviation in x,y but I do not understand why z is off and shows a constant velocity.
@mhkabir I tried to use INAV or LPE, actually spent quite some time on it, but I did nit get it working. (Disabling INAV (thought was needed to enable LPE), at least in the version I tried, lead to system which did not boot anymore). Can you point me in the right direction? 
@ecmnet  Thanks for pointing me to that issue. Could be that this issue is related to that issue I saw but it seems there are several things I have to look into to be sure.

mhkabir, 2016-01-29:
You *should* try LPE now. I regularly fly this, as do others, so it's stable : #3209 . It is guaranteed to perform better than your current method.

Supporting a non-standard/experimental setup like yours is quite hard since there are a multitude of things that could be going wrong.

Thomas10111, 2016-01-29:
Ok. Thanks we will try LPE.

closed 2017-04-02 by mhkabir