PX4/PX4-Autopilot #3629: UAVCAN interfaced external sensors broken

mhkabir, 2016-01-29:
Some changes in commander seem to prevent arming when external sensors come online late. I'm currently tracking this down.

QGC displays the error message "Preflight checks complete, reboot before arming." A reboot, obviously, doesn't fix the issue.

@pavel-kirienko

mhkabir, 2016-01-29:
Looks like we never get out of ```ARMING_STATE_STANDBY_ERROR``` although sensor comes online later.

LorenzMeier, 2016-02-01:
Then extending the hot plug timeout should be enough to fix it.

pavel-kirienko, 2016-02-05:
Not reproducible on latest master. The testing setup included 4 Zubax Orel ESC and 1 Zubax GNSS 2. I tried to re-configure and recalibrate the vehicle from scratch (wiped out all parameters), everything worked smoothly, at the end PX4 can arm successfully.

pavel-kirienko, 2016-02-08:
Closing because the problem was caused by bad cables.

mhkabir, 2016-02-10:
@pavel-kirienko @LorenzMeier OK, so this was not cables. Reopening. 

I have zero errors on the CAN interface now; Issue still occurs for me on a fresh system, recalibrated and everything.

```
[ERROR] [1455095525.661711420]: FCU: Preflight check resolved, reboot before arming
```
Cannot arm.

mhkabir, 2016-02-10:
Okay, I figured out the issue. 
This was not even related to UAVCAN. The system had low battery, so it didn't go to standby. We probably need better reporting here, but not related to this. Closing.

closed 2016-02-10 by mhkabir