PX4/PX4-Autopilot #3630: An error when i make px4fmu-v2_default in terminal with ubuntu.A newer needs help!

JohnPanheixia, 2016-01-29:
in terminal:
wybbd@wybbd-virtual-machine:~/src/Firmware$ make px4fmu-v2_default
git submodule init
Tools/check_submodules.sh 
Git submodule config valid.
Checked NuttX submodule, correct version found
Checked Tools/gencpp submodule, correct version found
Checked Tools/genmsg submodule, correct version found
Checked Tools/jMAVSim submodule, correct version found
Checked Tools/sitl_gazebo submodule, correct version found
Checked cmake/cmake_hexagon submodule, correct version found
Checked mavlink/include/mavlink/v1.0 submodule, correct version found
Checked src/lib/DriverFramework submodule, correct version found
Checked src/lib/dspal submodule, correct version found
Checked src/lib/ecl submodule, correct version found
Checked src/lib/matrix submodule, correct version found
Checked src/modules/uavcan/libuavcan submodule, correct version found
Checked unittests/googletest submodule, correct version found
make -C /home/wybbd/src/Firmware/build_px4fmu-v2_default -j4 --no-print-directory 
make[1]: *** 没有指明目标并且找不到(no direction and could not find) makefile。 停止(stop)。
make: *** [px4fmu-v2_default] 错误(error) 2
----------------------------------------------------------------------------------------------
or in terminal:
wybbd@wybbd-virtual-machine:~/src/Firmware$ sudo make px4fmu-v2_default
[sudo] password for wybbd: 
git submodule init
Tools/check_submodules.sh 
Git submodule config valid.
Checked NuttX submodule, correct version found
Checked Tools/gencpp submodule, correct version found
Checked Tools/genmsg submodule, correct version found
Checked Tools/jMAVSim submodule, correct version found
Checked Tools/sitl_gazebo submodule, correct version found
Checked cmake/cmake_hexagon submodule, correct version found
Checked mavlink/include/mavlink/v1.0 submodule, correct version found
Checked src/lib/DriverFramework submodule, correct version found
Checked src/lib/dspal submodule, correct version found
Checked src/lib/ecl submodule, correct version found
Checked src/lib/matrix submodule, correct version found
Checked src/modules/uavcan/libuavcan submodule, correct version found
Checked unittests/googletest submodule, correct version found
为 'NuttX' 同步子模组 url
为 'Tools/gencpp' 同步子模组 url
为 'Tools/genmsg' 同步子模组 url
为 'Tools/jMAVSim' 同步子模组 url
为 'Tools/sitl_gazebo' 同步子模组 url
为 'cmake/cmake_hexagon' 同步子模组 url
为 'mavlink/include/mavlink/v1.0' 同步子模组 url
为 'src/lib/DriverFramework' 同步子模组 url
为 'src/lib/dspal' 同步子模组 url
为 'src/lib/ecl' 同步子模组 url
为 'src/lib/matrix' 同步子模组 url
为 'src/modules/uavcan/libuavcan' 同步子模组 url
为 'unittests/googletest' 同步子模组 url
CMake Error: The source directory "/" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** [px4fmu-v2_default] 错误 1

The problem has confused me for the whole day!

kd0aij, 2016-01-29:
If you want to post an issue, you should at least make the effort to include all of the steps necessary to reproduce it.

davids5, 2016-01-29:
1) Why the sudo?
2) use ```git submodule update --init --recursive```
3) ```make px4fmu-v2_default```

JohnPanheixia, 2016-01-30:
3Q,it has been solved

closed 2016-01-30 by JohnPanheixia

JarryChou, 2016-03-08:
How did you solve this ? Please describe in detail ,thanks a lot.

JohnPanheixia, 2016-03-08:
@JarryChou do this on the page       http://dev.px4.io/starting-installing-linux.html