PX4/PX4-Autopilot #3635: The mc_att_control attitude_ sp and thrust_sp output missed serveal times in the fly test.

CornerOfSkyline, 2016-01-31:
I test the standard vtol auto mission. I found that the mc_att_control attitude_ sp and thrust_sp output missed serveal times in the fly test. The motors‘ rotational speed was the armed speed.

The log file is here.

http://logs.uaventure.com/view/psa25rfC5NmgC7EcGqxNM3

AndreasAntener, 2016-01-31:
Are you talking about the following 0 thrust events during the landing phase?
[image]

CornerOfSkyline, 2016-01-31:
Yes, it happened during the landing phase.
@AndreasAntener

AndreasAntener, 2016-01-31:
The problem is that the downwards velocity suddenly gets close to 0 so it thinks it's on the ground and cuts thrust. Pure multicopters have a more constant velocity during landing than VTOLs. In wind they tend to "stop" in air during the landing because the wings generate lift. I'd recommend to try 2 things:
- slightly higher landing speed: MPC_LAND_SPEED to 0.8
- improve Z velocity tuning, for instance try increasing MPC_Z_VEL_P a bit

closed 2016-01-31 by AndreasAntener