PX4/PX4-Autopilot #3638: Acceptance radius confusion for user

tumbili, 2016-01-31:
There is a radius in the mission item and there is a parameter. This is confusing for the user and we have to re-check what the system actually does. During our test we had the feeling that the plane was ignoring the parameter. Logs to come.

LorenzMeier, 2016-01-31:
For planes the radius is only a maximum: It will choose the minimum it can fly or this one. But if you set it to 10m and it can only turn in 60m it will switch at 60m.

LorenzMeier, 2016-01-31:
Generally this should not be shown to users. They should just draw lines.

tumbili, 2016-01-31:
@LorenzMeier Is this the loiter radius or what?

LorenzMeier, 2016-01-31:
I can explain tomorrow on the whiteboard how the navigation works. But fundamentally you do not get to choose as a user when it switches the waypoint.

LorenzMeier, 2016-02-06:
This will be fixed in:
https://github.com/mavlink/qgroundcontrol/issues/2743
and
https://github.com/mavlink/qgroundcontrol/issues/2744

LorenzMeier, 2016-02-17:
Fixed in QGC.

closed 2016-02-17 by LorenzMeier