PX4/PX4-Autopilot #3639: Plane ignoring waypoints

tumbili, 2016-01-31:
We have a mission where the plane just ignored a waypoint (I also saw the double jump of mission index in QGC) and from then on the mission didn't make any sense anymore. It actually tried to land in an arbitrary spot, although there was a landing waypoint set. Logs, mission file and pictures from the QGC screen to come.

tumbili, 2016-01-31:
We could reproduce this about 3 times consequently.

tumbili, 2016-02-02:
Mission:
´´´
QGC WPL 120

1	1	3	22	0.174532920122146606	1.40129846432481707e-45	0	0	47.4285507202148438	8.41064548492431641	40	1

2	0	3	16	0	3	0	0	47.4283714294433594	8.41195964813232422	250	1

3	0	3	16	0	3	10	0	47.4298248291015625	8.41264057159423828	60	1

4	0	3	19	30	3	10	0	47.4305686950683594	8.41049480438232422	60	1

5	0	3	16	0	3	0	0	47.430419921875	8.4116058349609375	20	1

6	0	3	16	0	3	0	0	47.4287376403808594	8.41081142425537109	20	1

7	0	3	16	0	3	0	0	47.4283256530761719	8.41153430938720703	40	1

8	0	3	16	0	3	0	0	47.4313125610351562	8.41213226318359375	40	1

9	0	3	16	0	3	0	0	47.4306793212890625	8.41280841827392578	40	1

10	0	3	21	25	3	0	0	47.428985595703125	8.41188621520996094	0	1

´´´

tumbili, 2016-02-02:
http://logs.uaventure.com/view/NuRsf9vCRpnBPszDFcxepY

This is a mission where the landing did not work at all. You can see in the map that we switched out of mission mode where the path of the plane changes colour to red (plane was heading south east).
It should have approached the landing waypoint which a southern heading but possibly since the last two wapoints were too close to each other it aligned differently (mission index jumped by two!).
This is a similar problem that you have when trying to land again after an aborted mission.

tumbili, 2016-02-02:
This is a mission where the landing did not work at all. You can see in the map that we switched out of mission mode where the path of the plane changes colour to red (plane was heading south east).
It should have approached the landing waypoint which a southern heading but possibly since the last two wapoints were too close to each other it aligned differently (mission index jumped by two!).
This is a similar problem that you have when trying to land again after an aborted mission.

tumbili, 2016-02-02:
http://logs.uaventure.com/view/NuRsf9vCRpnBPszDFcxepY

LorenzMeier, 2016-02-09:
The landing waypoint check was disabled. You would now not be allowed any more to set the mission like you did.

closed 2016-02-09 by LorenzMeier