PX4/PX4-Autopilot #3642: SITL gazebo simulation question

tuofeichen, 2016-01-31:
Hey guys,
Noob here to PX4 and just setup the sitl gazebo simulator using the make posix_sitl_default in the Firmware. Get the communication going and everything works fine. I'm using a small offboard ROS node with mavros plugin setpoint_velocity to send setpoint_velocity plugin to input velocity command but what I'm wondering is, 

Why does the drone drift a lot when I'm only sending a upward z direction velocity? Is the gazebo model ideal with no wind? Or is there anything about how px4 low level control/setpoint_velocity command I'm not aware of?

Thanks

Best Regards,
Francis

gokoreas, 2016-02-05:
The drift maybe due to the barometer data

To reduce it, changing the parameter of the INAV weights may change how the vehicle will move

I guess if you set the setpoint_velocity as you said will not necessarily control it self to stay around a fixed point and may drift

you will have to control the quad or use the setpoint_position to stay it in a fixed point

LorenzMeier, 2016-02-05:
@tuofeichen @gokoreas Are you using master?

tuofeichen, 2016-02-05:
Hi @LorenzMeier 

Yeah, the Firmware and SITL I'm using is master. I also have a feeling that my computer is too slow to run gazebo, because when I ran jmavsim, it works most of the time. I was using a virtual machine and recently I installed the dualboot and I will see if things get better.

Still, would appreciate you explain a little bit how px4 handle the setpoint_velocity command, and how does the IMU estimate velocity. Because I did see the message published from `mavros /local_position/velocity` and I wonder if that's usually reliable. 

Thanks

tuofeichen, 2016-02-08:
Update: stream velocity not fast enough causes the problem. Already solved.

andre-nguyen, 2016-02-08:
How fast are you sending the setpoints now?

tuofeichen, 2016-02-08:
sending it every 10ms should be fine, careless mistake on my side, set the `ros::Rate loop_rate` to be 10 originally, seeing that 10ms in documentations. And never thought about that again. Setting it to 100 (hz) solved the problem.

andre-nguyen, 2016-02-09:
Interesting, I'll try adding that tip in the documentation when I can.

closed 2016-04-01 by LorenzMeier