PX4/PX4-Autopilot #3645: Extend camera trigger to distance based mode and expose in QGC

LorenzMeier, 2016-02-01:


NaterGator, 2016-02-17:
Percent forward overlap is another useful triggering mode but requires knowledge of the camera parameters and distance to terrain.

If you can make the assumption that the camera position is normally nadir and the terrain is flat the problem simplifies to roughly distance based triggering with a little more robustness.  It may be worth implementing at the same time.

birchera, 2016-02-17:
An implementation of the distance based triggering has been proposed in https://github.com/PX4/Firmware/pull/3774

The distance can be adapted in QGC.

In my opinion handling the overlap and camera parameters like e.g. FoV, resolution would rather belong in a mission planning feature in QGC. From these parameters you could then derive the value for the triggering distance parameter.

LorenzMeier, 2016-03-13:
Done!

closed 2016-03-13 by LorenzMeier