PX4/PX4-Autopilot #3655: Landing "kangaroo"

AndreasAntener, 2016-02-02:
@tumbili Basically thr_max is 0, it's sitting on the ground and practically not moving, landing type setpoint is still active, but then suddenly is allowed to throttle up again. As far as I can tell it's not the free fall detection triggering it.
http://logs.uaventure.com/view/xmkioKbqXpQfcNqBsTpebX
http://logs.uaventure.com/view/c2erHVJtUvqdhfRv7CBUUk

sanderux, 2016-02-05:
This behavior can be seen in this video at the end: https://drive.google.com/open?id=0B27AvDmVit4KRVlpYkFMNVpTLWM

tumbili, 2016-02-08:
@AndreasAntener @sanderux The problem could be this one
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1533
If this condition is not satisfied then the max thrust parameter will not be set to zero any more but rather to standard max thrust value a few lines up in the code. The velocity at / after touchdown is anyway not really accurate any more due to the impact. Will have a closer look tomorrow.

tumbili, 2016-02-10:
Adressed in #3716

PR #3716 (2016-02-16): Kangaroo fix
Possibly fixes #3655
@AndreasAntener Can you have a quick look? After we assume to have reached the ground a flag is set which we use to always cut throttle when this flag was set true. Only the free fall detection can revoke this.
I would check the first commit for the diff, the other one is just code style
closed 2016-02-16 by AndreasAntener