PX4/PX4-Autopilot #3663: VTOL: RTL on aborted transitions in auto

AndreasAntener, 2016-02-03:


sanderux, 2016-02-04:
For my own reference:
format command
https://github.com/PX4/Firmware/blob/15b72045ceda4db2bc76c1fb2366cdb6a55d02d3/src/modules/commander/commander.cpp#L804-L807
issue command
https://github.com/PX4/Firmware/blob/15b72045ceda4db2bc76c1fb2366cdb6a55d02d3/src/modules/navigator/mission_block.cpp#L113-L117
here (only mission mode?)
https://github.com/PX4/Firmware/blob/bde7eea4388a8017031da5c89c313ec18adddca4/src/modules/vtol_att_control/vtol_att_control_main.cpp#L455

sanderux, 2016-02-04:
@AndreasAntener @LorenzMeier do you think the failsafe should trigger flight termination or just switch vehicle_status to MAIN_STATE_AUTO_RTL?

LorenzMeier, 2016-02-04:
I don't think you propose to crash the plane if the transition failed, do you?

sanderux, 2016-02-04:
As i read it, flight termination has param2==1 for RTL :)

sanderux, 2016-02-04:
I think it should consider failsafe procedures as transition from manual mode might fail with gps lock. Or should failsafe only be triggered in mission mode?

PR #3719 (2016-02-10): Vtol transition failsafe with RTL for standard VTOL
@sanderux 2 additional fixes
- reset transition command when not in any of the auto modes (missed that on my last push)
- reset failsafe state after switched out. otherwise you cannot engage mission again

Partly replaces #3676
Fixes: #3663
closed 2016-02-10 by AndreasAntener