PX4/PX4-Autopilot #3666: Lidar Lite I2C Interface Both On and Off?

brandonr49, 2016-02-03:
Lidar lite over i2c gives conflicting messages about whether or not the i2c drivers have started.

nsh> ll40ls start i2c
LL40LS on I2C bus 1 at 0x62 (bus: 100 KHz, max: 100 KHz)
nsh> ll40ls test i2c
ll40ls: /dev/ll40ls_int open failed, is the driver running?: No such file or directory
nsh> ll40ls start i2c
ll40ls: driver already started

Is the driver initializing incorrectly?

SimonWilks, 2016-02-03:
This driver is probably running but when the test runs it's just taking a guess (the message that is), it's probably not really checking. The problem is the connection isn't there so it can't find the lidar.

Which version of the lidar are you using? Are you powering it sufficiently (you may need to use a BEC for I2C)? Are you using the one with the blue label? Have you considered using the PWM interface instead?

brandonr49, 2016-02-03:
Hey, thanks for the quick response. 

I'm using the BEC onboard the Pixracer (I'm plugging into the 5v out for peripherals) and I am using the blue label Lidar Lite. I was using it on PWM before now with a Pixhawk but the Pixracer doesn't support PWM so I'm trying to switch to i2c. 

So you think the message means that the connection is bad but that the driver probably started? Doesn't the connection have to exist for this "LL40LS on I2C bus 1 at 0x62 (bus: 100 KHz, max: 100 KHz)" to come back?

SimonWilks, 2016-02-03:
A quick look in the code shows this message is generated when it can't open the I2C connection. The test function tries to open it itself. All I can suggest is try powering the lidar externally (see http://planner.ardupilot.com/wp-content/uploads/sites/5/2015/04/RangeFinder_LIDARLite_Pixhawk_v2.png) and otherwise see if anyone knows if there is something pixracer specific issue with I2C and the lidar.

kd0aij, 2016-02-03:
This might be similar to the problem with the 3DR digital airspeed sensor
on PixRacer: if the I2C interface of the LidarLite isn't running on 3.3V,
there may be bus errors. This is due to the fact that the STM32F4 I2C
module runs on 3.3V, and its logic high level is marginal for a 5V I2C
slave.

brandonr49, 2016-02-04:
Thanks for the help.
Simon was right and the driver was running; I just needed to add debug messages to see the data coming through.

This "issue" can be closed.

LorenzMeier, 2016-02-04:
@brandonr49 Can you send a pull request for your fixes?

brandonr49, 2016-02-05:
Will do this weekend. I made a lot of changes that I definitely shouldn't commit.

LorenzMeier, 2016-02-18:
@brandonr49 Any chance for the PR?

LorenzMeier, 2016-07-28:
@brandonr49 How is your PR looking?

devbharat, 2016-10-31:
This is still the case with Lidarlite v3. The driver runs fine and publishes the messages, but the 'test' on i2c says `dev/ll40ls_int open failed, is the driver running?`

cjepple, 2016-11-13:
I've got a couple related issues with the LidarLite 3 and I2c. Using build 1.5.0dev. The ll40ls pwm starts automatically, so in nsh I stop the ll40ls pwn version, and start the o2c version. It shows up as /dec/ll40ls_ext I can run a test with -X and I get back valid data and passes test. However, when I look at Maclink Inspector, I'd expect to see the distance sensor displaying the lidar data, but there is no distance sensor shown. If I set it up for pwm, itallworks fine.

So it looks like i2c is working when I start it manually and do a test, but doesn't show up in Mavlink inspector, so I assume it is not going into the solution.

As a separate issue, I'm trying to modify the startup ./init.d/rcS to start the ll40ls i2c driver and not the pwm version. How is this done on-board? nsh doesn't have vi, and I can't upload a file via QGroundControl, and it doesn't appear on the SD card. Do I need to reload the entire firmware just to change the startup script?

Thanks for any help. -Chris

LorenzMeier, 2016-12-28:
Fixed in 381611bb66. These are all usage errors, running the commands with the full arguments would have worked. The confusion arises from some commands offering convenience defaults while others not.

closed 2016-12-28 by LorenzMeier

kd0aij, 2016-12-28:
@cjepple In answer to "Do I need to reload the entire firmware just to change the startup script?"
Look at http://dev.px4.io/advanced-system-startup.html
/etc/extras.txt may allow you to do what you want without modifying rcS

cjepple, 2016-12-28:
I had finally come across ./extras.txt and that did exactly what I needed. Thanks!