PX4/PX4-Autopilot #3683: LPE X-Y Position estimation based on Flow not in correct units

ecmnet, 2016-02-06:
Although position hold is working basically, it seems as if the absolute units LPE estimates for local position in XY direction are not correct. Having moved the vehicle by about 1m, LPE shows a movement of about 0.2m. 

[http://logs.uaventure.com/view/RBzbbkjLCfPtU69py7w5QQ#Local_X_PLOT](url)

To achieve better position hold results, I need to tune PIDs, but I would like to do that based on correct estimations.

ecmnet, 2016-02-06:
Note that issue also below: The chart shows a max. positioning error of about 0,15m, while the vehicle moved nearly 1m:

[image]

mhkabir, 2016-02-06:
There might be a need for a empirical scaling factor as used in inav and EKF-1. 

Also, what rangefinder are you using?

mhkabir, 2016-02-06:
Also, note that the LocalX and LocalY are NED frame estimates. So, they're decomposed. Are you sure that you moved 1m aligned to e.g North?

mhkabir, 2016-02-06:
Estimator seems to be faulting continuously as well. See the last estimator health plot. Are you using the PX4Flow master firmware?

ecmnet, 2016-02-06:
@mhkabir ad1: I am using a LaserRangeFinder with 1cm resolution (SF10A), which works perfect. Also altitude hold is ok. ad2: I am aware of that. Manual movements exactly to the North show similar results.
ad3: Yes, this is the PX4Flow master. Ground has a good contrast, although this is indoors (btw. I also have had an ADNS3080 working in the same environment which resulted in good flow quality).

ecmnet, 2016-02-06:
@mhkabir Strange: When I look to the raw flow quality, it looks like this:

[image]
:
This should be sufficient and there are no drop-outs.

ecmnet, 2016-02-07:
Additionnally, I found out that the scale in Y (EW) direction differs from Y (NS). Think I have to check Px4Flow...

mhkabir, 2016-02-07:
Yeah, that can happen. EKF1 had separate FX_SCALER and FY_SCALER params, IIRC.

ecmnet, 2016-02-07:
I am just digging into the PX4Flow Firmware. There I saw that the integration of flow is only performed when the sonar has valid measurements. On the other hand, BlockLocalPositionEstimator takes flow as provided by PX4Flow (and does not compensate with published altitude). Am I right?

ecmnet, 2016-02-07:
Just removed the sonar compensation in PX4Flow (=assume altitude 1m). Flying in that height approximately gives now correct LPE scales. So for users that don't want to rely on sonar (indoors you have e.g. carpets or reflexions) but on laser range finders, I would suggest

a) add a parameter to PX4Flow to disable sonar compensation (or sonar at all?)
b) add a parameter to LPE to compensate flow by altitude within LPE

What do you think?

mhkabir, 2016-02-07:
> BlockLocalPositionEstimator takes flow as provided by PX4Flow (and does not compensate with published altitude). Am I right?

You are not. https://github.com/PX4/Firmware/pull/3209/files#diff-478580be4572c29aaaa7e5861078259eR905

mhkabir, 2016-02-07:
I will add a little something to disable Sonar when a Lidar is present. Currently, both are used in the update loop, which might be causing a bad Z-estimate.

mhkabir, 2016-02-07:
> add a parameter to PX4Flow to disable sonar compensation (or sonar at all?)

Flow firmware doesn't use the sonar to compensate anything. It just integrates flow measurements if the sonar is within range (which makes sense since the lens has a narrow focus band)

ecmnet, 2016-02-07:
You're right. PX4Flow compensates only for speed estimation, not for accumulated flow. Still does it make no sense to me to relay on sonar for integration, when more accurate laser range data is available.
I think my issue is, that (in my case) indoors the sonar is not working properly all the time.

I don't think I am facing a wrong Z estimation (although sonar AND Lidar is present) - as soon as Lidar is initialized by LPE, the altitude is perfect.

Regarding LPE flow altitude compensation: I don't get it right now. In correctSonar/Lidar I see clearly Z axis updates on the Kalman, but no flow altitude compensation...

mhkabir, 2016-02-07:
> no sense to me to rely on sonar for integration,

As I've reiterated, multiple times, sonar DOES NOT affect flow integration on the PX4Flow side. Just that the flow is accumulated if sonar distance > 30 cm. The lens is out of focus closer than that anyway.

> I think my issue is, that (in my case) indoors the sonar is not working properly all the time

What makes you think this? Can you try to compare the sonar values and lidar distance values in the logs?

ecmnet, 2016-02-07:
Just did a test-flight in posHold mode (with disabled distance check in flow): Still it seems to me that position estimation is too low (aprox. by factor 3-4):

[image]

Flow quality was good at all time:

[image]

Altitude was about 0.7m.

mhkabir, 2016-02-07:
Try putting a 1.35x (or whatever )factor multiplier here and try it out : https://github.com/PX4/Firmware/pull/3209/files#diff-478580be4572c29aaaa7e5861078259eR902

You can also try disabling/removing the gyro compensation.

ecmnet, 2016-02-07:
Yes, as I've agreed before, Flow does NOT compensate flow integration (though it does for speed estimation, which is not used). But it is clear that the distance check in Flow does affect the flow massively in my surrounding (carpets). So I will stay with disabled check for now.

Could you share some numbers, what is the achievable accuracy in poshold with LPE in your experience?

mhkabir, 2016-02-07:
> Could you share some numbers, what is the achievable accuracy in poshold with LPE in your experience?

I've had a pretty good experience with LPE sometime back (~2 months) on a small 250-size quadrotor. The gains were properly tuned, and the drift was fairly low, well under 10cm for about 3 mins of flight. I'll see if I can find those logs for you, or do another flight after reassembling the copter. 

Something you can do to help would be to stick a GPS unit on the quad and send me the logs of an outdoor flight with flow + GPS. Then I can compare the estimated velocities from GPS and Flow. A VICON or other motion capture environment (if you have access to one) for groundtruth would be even more ideal.

mhkabir, 2016-02-07:
Also, can you give me the log to that last flight of yours? Please put it up on LogMuncher.

ecmnet, 2016-02-07:
Here is the log:

http://logs.uaventure.com/view/Wz2eijS7JFjtJsxY8UKPLL

As soon as the weather allows outdoor testing, I will try to compare velocities.

mhkabir, 2016-02-07:
OK, it's weird that the estimator keeps faulting for optical flow. Can you try putting in a few printfs or mavlink_log()'s and see what's going on?

mhkabir, 2016-02-07:
Remove this line here and try a flight : https://github.com/PX4/Firmware/pull/3209/files#diff-478580be4572c29aaaa7e5861078259eR989

ecmnet, 2016-02-07:
Yes, I also noticed that. Ok, I will give that a try.

ecmnet, 2016-02-07:
First dry-moves show that (as expected), flow updates are now regularly. I'll try a flight by tomorrow.

ecmnet, 2016-02-08:
@mhkabir Did a test-flight: http://logs.uaventure.com/view/zgUWwccUk4zK8tANynHnfA. Estimation seem to be better with disabled flow-validity check, so you assumption is right. But even with this modification, I see an underestimation by aprox. 2. (Those two posHold points shown below were about 1m from each other):

[image]

ecmnet, 2016-02-08:
Another test-flight, where I ensured that Loc.Pos.Z represent distance to ground. Position estimation looks correct now (did posHold with some manual changes):

[image]

mhkabir, 2016-02-08:
> How to ensure that altitude compensation is always done by distance to ground

It's *always* done by distance to ground if your rangefinder is working properly.

> Why is LPE assuming that the flow data is invalid?

See the errors which QGC shows you.

ecmnet, 2016-02-08:
QGC messages don't help here (saying flow invalid), because that seems not to be correct. (Otherwise the patch wouldn't work).

mhkabir, 2016-02-08:
I mean, the *exact* message. You see this : ```[lpe] bad flow data``` ?

ecmnet, 2016-02-08:
I disabled the flow-quality check and compensated flow directly by lidar distance (if valid). This solved my issues and posHold accuracy now is about 0.15m (untuned). For now I am closing the issue. Thank you again.

closed 2016-02-08 by ecmnet