PX4/PX4-Autopilot #3684: Takeoff in mission: Vehicle rises unlimited

LorenzMeier, 2016-02-06:
@AndreasAntener @sanderux @tumbili If I take off in SITL from the ground using a mission (NO GAMEPAD! SWITCH TO MISSION, THEN ARM!) the vehicle raises uncontrolled and never switches to to flying to the actual waypoint.

AndreasAntener, 2016-02-06:
The mission item here https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission_block.cpp#L65 seems to be initialized with nan values in SITL, but not on hardware. This breaks the takeoff check. I changed this in the mission update and will make a PR to master in a sec.

LorenzMeier, 2016-02-06:
The zero might be an issue. The braces should be empty.

AndreasAntener, 2016-02-06:
Same problem with empty braces. But there is definitely something wrong with the init, or the memcpy from mission reading. I get different random values on different branches (2 branches where the mission item struct is different).

For example: next item 22, rel 1, lat 47.3669, lon 8.5498, alt 25.00, yaw 0.00, loirad 0.00, loidir -1, rad nan, tin 212925686757085407282649366528.00, pmin 0.26, auto 1, origin 0, jmi 2121728, jrc -742746624, jcc 0, p1 0.00, p2 0.00, p3 -0.00, p4 0.00, p5 0.00, p6 0.00, p7 0.00

sanderux, 2016-02-06:
is https://github.com/PX4/Firmware/issues/3685 related to this?
i get different values depending on what sitl model i use (gazebo or gazebo_standard_vtol)

sanderux, 2016-02-06:
On current master i can not reproduce this. I have tried; 
- make posix_sitl_default jmavsim
- make posix_sitl_default gazebo
- make posix_sitl_default gazebo_standard_vtol
- make posix_sitl_default gazebo_tailsitter

I'm on gcc (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4

AndreasAntener, 2016-02-07:
Is this maybe just related to OSX and struct alignment?

Debugging mission items with master gives me the following values after reading the first item:
```
WARN  next item 22, rel 1, lat 47.3669, lon 8.5498, alt 25.00, yaw 0.00, loirad 0.00, loidir -1, rad nan, tin 0.00, pmin 0.26, auto 1, origin 0, jmi 2121728, jrc 174511104, jcc 0, p1 0.00, p2 0.00, p3 -0.00, p4 0.00, p5 0.00, p6 0.00, p7 0.00
```

Note the "rad nan"

After changing the mission_item_s struct (using different datatypes), I get the following output for the same mission item after reading:
```
WARN  next item 22, rel 1, lat 47.3669, lon 8.5498, alt 25.00, yaw 0.00, loirad 0.00, loidir -1, rad -39614680997675493241714114560.00, tin 0.00, pmin 0.26, auto 1, origin 0, jmi 816840704, jrc -1845379908, jcc 0, p1 0.00, p2 -0.01, p3 0.00, p4 0.00, p5 0.00, p6 0.00, p7 0.00
```

(I deleted the dataman file in between)

LorenzMeier, 2016-02-07:
Fixed on master.

closed 2016-02-07 by LorenzMeier