PX4/PX4-Autopilot #3685: sitl gazebo_standard_vtol bool init issue

sanderux, 2016-02-06:
It seems a newly added bool to the attitude setpoint struct gets random values in gazebo_standard_vtol as is it were not initialized.

other sitl makes seem to not have this problem.

Test can be done by running: https://github.com/PX4/Firmware/compare/vtol_weathervane

sanderux, 2016-02-06:
make posix_sitl_default gazebo
[image]

make posix_sitl_default gazebo_standard_vtol
[image]

AndreasAntener, 2016-02-07:
Can you check if the problem still exists if you initialize the attitude setpoint in the position controller with {} instead of the memset?

AndreasAntener, 2016-02-07:
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L386

LorenzMeier, 2016-02-07:
@sanderux @AndreasAntener The fix would need to be in the controller publishing the attitude setpoint - in this case the VTOL controller.

AndreasAntener, 2016-02-07:
Right, it needs to be fixed in the pos and vtol controller. All vtol types currently memcpy the setpoint in MC mode.

sanderux, 2016-02-07:
@AndreasAntener @LorenzMeier 

The problem turned out to be the fact that the virtual att setpoints in the vtol ctl have their own structs:
https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/vtol_type.h#L95-L97
https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/vtol_att_control_main.h#L164-L166
 
they get copied to the standard vehicle_attitude_sp here:
https://github.com/PX4/Firmware/blob/f41f60901db7c62c2a06104fd0a2480b85bb95d8/src/modules/vtol_att_control/standard.cpp#L258

Those structs are not exact copies of vehicle_attitude_setpoint_s. 
https://github.com/PX4/Firmware/blob/master/msg/vehicle_attitude_setpoint.msg
https://github.com/PX4/Firmware/blob/master/msg/fw_virtual_attitude_setpoint.msg
https://github.com/PX4/Firmware/blob/master/msg/mc_virtual_attitude_setpoint.msg

I don't know why these are not just instances of vehicle_attitude_setpoint_s.

I would propose changing this so these 2 virtual structs become instances of the standard vehicle_attitude_setpoint_s to avoid future complications.

Do you see any reason why these should have their own structs?

tumbili, 2016-02-07:
@sanderux I agree it's better to have one message for all of them. Otherwise if somebody changes one of the topics (like you did) and does not alter the other ones things will not work anymore.
However, I'm not sure if this is supported, will need to check.

sanderux, 2016-02-07:
@tumbili 
I have a working version of this in SITL. I see no strange behavior.
I will push to https://github.com/PX4/Firmware/tree/vtol_weathervane so you can check

sanderux, 2016-02-07:
Resolved 
https://github.com/PX4/Firmware/pull/3694

closed 2016-02-07 by sanderux