PX4/PX4-Autopilot #3700: Sample Mode

robinlee09201, 2016-02-07:
APM has a flight mode for copters called "Sample mode", which allows the pilot to control a copter regardless its heading. Basically, the copter will take the heading when it's armed as the front direction, though the real heading may change during flying. It's extremely helpful for the beginners. This feather is also supported by Multii Wii Copter, known as "Head Free Mode". Do you have any interest to introduce this feature into px4's firmware?

sanderux, 2016-02-07:
I believe that's called "Simple mode" but that might have been a typo

robinlee09201, 2016-02-08:
OOPS, true

dagar, 2017-04-11:
Is anyone still interested in this? I think it would be fairly easy to implement as a position control submode, especially once we refactor the manual setpoint handling into stick mapper.

dagar, 2017-04-28:
I'll consider this once we have the stick mapper.

closed 2018-02-21 by PX4BuildBot