PX4/PX4-Autopilot #3702: Disable (yaw based) thrust reduction for VTOLs

sanderux, 2016-02-08:
This has a negative effect on VTOLs
https://github.com/PX4/Firmware/blob/6a98c6ca630a1fcacdf4c24bfec9d0b2c65201f9/src/modules/systemlib/mixer/mixer_multirotor.cpp#L338

LorenzMeier, 2016-02-08:
If you hit that your system is likely just too underpowered. You can't have the cake and eat it. It would mean you have no yaw authority at all and your attitude control is marginal.

sanderux, 2016-02-08:
@LorenzMeier 
I am currently 2.4:1 powered. I tried 40kph crosswind on the tail (no config would be able to hold that). the priority at that point seemed to go to maintaining yaw in favor of altitude. at 40kph front wind it was hovering nicely at 60% throttle

sanderux, 2016-02-09:
here's a log of manual hover and loiter with nose in 40kph wind: http://logs.uaventure.com/view/YRzDvtPrWutTwj4vgdJT4R#AS_Pitch_PLOT

here's a log of the same 40kph wings while i turn the vehicle sideways to the wind (cross wind): http://logs.uaventure.com/view/ysBvQKS4jAfkxYLqA8TSWd

sanderux, 2016-02-23:
Solved by yaw timeout and weathervane

closed 2016-02-23 by sanderux