PX4/PX4-Autopilot #3705: SITL Param dump

tuofeichen, 2016-02-08:
Hey guys,
I was wondering when I do a `mavros param dump` when running SITL, the parameter lists don't seem to be complete. Specifically, it seems like its not dumping all the INAV parameter, altitude Q estimator and so forth. Whereas when I actually run it on px4, the dumped parameter included all those. 

Looking at the build folder, those parameters are defined in 'parameters.xml', is there a reason behind its not dumping it in mavros? (maybe not want us to change those parameters?) Because I was trying to load some of my hardware parameter into SITL just to see what happens (for instance INAV ones), and mavros node output a warning saying those parameters not defined. 

Thank you
SITL parameters:
`
#NOTE: 06.02.2016 12:19:33
BAT_CAPACITY,-1.0
BAT_C_SCALING,-1.0
BAT_N_CELLS,3
BAT_V_CHARGED,4.199999809265137
BAT_V_EMPTY,3.4000000953674316
BAT_V_LOAD_DROP,0.07000000029802322
BAT_V_SCALING,-1.0
CAL_ACC0_ID,1376264
CAL_ACC0_XOFF,0.009999999776482582
CAL_ACC0_XSCALE,1.0099999904632568
CAL_ACC0_YOFF,-0.009999999776482582
CAL_ACC0_YSCALE,1.0099999904632568
CAL_ACC0_ZOFF,0.009999999776482582
CAL_ACC0_ZSCALE,1.0099999904632568
CAL_ACC1_ID,1310728
CAL_ACC1_XOFF,0.009999999776482582
CAL_ACC1_XSCALE,1.0
CAL_ACC1_YOFF,0.0
CAL_ACC1_YSCALE,1.0
CAL_ACC1_ZOFF,0.0
CAL_ACC1_ZSCALE,1.0
CAL_ACC_PRIME,0
CAL_BARO_PRIME,0
CAL_GYRO0_ID,2293768
CAL_GYRO0_XOFF,0.009999999776482582
CAL_GYRO0_XSCALE,1.0
CAL_GYRO0_YOFF,0.0
CAL_GYRO0_YSCALE,1.0
CAL_GYRO0_ZOFF,0.0
CAL_GYRO0_ZSCALE,1.0
CAL_GYRO_PRIME,0
CAL_MAG0_ID,196616
CAL_MAG0_ROT,-1
CAL_MAG0_XOFF,0.009999999776482582
CAL_MAG0_XSCALE,1.0
CAL_MAG0_YOFF,0.0
CAL_MAG0_YSCALE,1.0
CAL_MAG0_ZOFF,0.0
CAL_MAG0_ZSCALE,1.0
CAL_MAG1_ID,0
CAL_MAG1_ROT,-1
CAL_MAG2_ID,0
CAL_MAG2_ROT,-1
CAL_MAG_PRIME,0
CBRK_AIRSPD_CHK,0
CBRK_ENGINEFAIL,284953
CBRK_GPSFAIL,240024
CBRK_RATE_CTRL,0
CBRK_SUPPLY_CHK,0
CBRK_USB_CHK,0
COM_AUTOS_PAR,1
COM_DISARM_LAND,0
COM_DL_LOSS_EN,0
COM_DL_LOSS_T,10
COM_DL_REG_T,0
COM_EF_C2T,5.0
COM_EF_THROT,0.5
COM_EF_TIME,10.0
COM_HOME_H_T,5.0
COM_HOME_V_T,10.0
COM_RC_IN_MODE,1
[... 642 lines cut ...]
SENS_BARO_QNH,1013.25
SENS_BOARD_ROT,0
SENS_BOARD_X_OFF,0.6722664833068848
SENS_BOARD_Y_OFF,-1.0588234663009644
SENS_BOARD_Z_OFF,0.0
SENS_DPRES_ANSC,0.0
SENS_DPRES_OFF,0.0
SENS_EN_LL40LS,0
SENS_FLOW_ROT,0
SYS_AUTOCONFIG,0
SYS_AUTOSTART,4011
SYS_COMPANION,921600
SYS_PARAM_VER,1
SYS_RESTART_TYPE,0
SYS_USE_IO,1
TRIG_MODE,0
TRIM_PITCH,0.0
TRIM_ROLL,0.0
TRIM_YAW,0.0
UAVCAN_ENABLE,0
VT_TYPE,0
`

Best
Francis

LorenzMeier, 2016-02-08:
SITL uses already the new estimator (EKF2). The programs you reference are not running - so its all working as it should. You need to change the start script in posix-configs/SITL if you like to use those instead.

closed 2016-02-08 by LorenzMeier

tuofeichen, 2016-02-08:
Hey Lorenz,
Thanks for the prompt reply. I guess I was asking a lot of this type of
questions because sometimes I'm not sure how px4 handles mavros in general.
And since I'm working with mavros for controlling the flight dynamics, is
there anywhere source-code-wise (in the px4 firmware I would imagine) I
should look into? In addition, I assume understanding mavlink is also going
to be helpful?

This is a good overall introduction, but I would like to go a little deeper
in terms of understanding.
https://pixhawk.org/dev/ros/mavros_offboard

This is not technically an issue so I just replied here instead of creating
a new one. Again, complete noob here and thanks for your help!

Best
Francis