PX4/PX4-Autopilot #3707: QGC/PX4 not communicating correctly

srees, 2016-02-09:
Behavior is similar to that caused on a linux system when https://github.com/PX4/Firmware/commit/e2fd2f466ebb5c2f8be0af0fbc2f2f8e77bf9c96 was implemented.

Fresh build of PX4 firmware on Ubuntu VM, latest QGC running on Mac host, different IP addresses.
QGC detects the launch of 'make posix gazebo' but immediately loses communication with it. Telem logs from the make command indicate connection is still active (closing/relaunching QGC is reflected with 'data link#0 lost/regained), but nothing is coming through to QGC. QGC reports that 'vehicle did not respond to request for parameters'

LorenzMeier, 2016-02-09:
I can reproduce this now here, should be a quick fix.

Inspirati, 2016-02-09:
i'm currently struggling with what i believe is much the same issue,
however i'm working with various archaic hardware (no VM's, not doing sim's),
looking forward to seeing suggestions to the fix i cannot seem to find...

LorenzMeier, 2016-02-09:
Ok, its fixed on master in 3e02bb1070b0f5f2909833ccb979ab4a03607d41. Please let me know if this now all works for you.

closed 2016-02-09 by LorenzMeier

srees, 2016-02-09:
Thank you, much better!