PX4/PX4-Autopilot #3708: unable to fetch in submodule path

sheenaagrawal, 2016-02-09:
I'm trying to build the PX4 flight stack firmware using the info given in this link : http://dev.px4.io/starting-building.html

I'm getting this error

```
Submodule 'jMAVlib' (https://github.com/PX4/jMAVSim.git) registered for path 'jM
AVlib'
fatal: unable to connect to github.com:
github.com[0: 192.30.252.129]: errno=No error

Unable to fetch in submodule path 'mavlink/include/mavlink/v1.0'
Checked NuttX submodule, correct version found
Checked Tools/gencpp submodule, correct version found
Checked Tools/genmsg submodule, correct version found
Checked Tools/jMAVSim submodule, correct version found
Checked Tools/sitl_gazebo submodule, correct version found
Checked cmake/cmake_hexagon submodule, correct version found
Checked mavlink/include/mavlink/v1.0 submodule, correct version found
Checked src/lib/DriverFramework submodule, correct version found
Checked src/lib/dspal submodule, correct version found
Checked src/lib/ecl submodule, correct version found
Checked src/lib/matrix submodule, correct version found
Checked src/modules/uavcan/libuavcan submodule, correct version found
Checked unittests/googletest submodule, correct version found
make -C /c/px4/Firmware/build_px4fmu-v2_default -j4 --no-print-directory
make[1]: *** No targets specified and no makefile found. Stop.
make: *** [px4fmu-v2_default] Error 2
```

LorenzMeier, 2016-02-09:
You or Github are having an internet connection issue. There is nothing wrong with the system.

closed 2016-02-09 by LorenzMeier

sheenaagrawal, 2016-02-09:
And what about **no target specified and no makefiles found**? @LorenzMeier

sheenaagrawal, 2016-02-09:
The error now is 

`Submodule 'jMAVlib' (https://github.com/PX4/jMAVSim.git) registered for path 'jM
AVlib'
fatal: Needed a single revision
Unable to find current revision in submodule path 'mavlink/include/mavlink/v1.0'

Checked NuttX submodule, correct version found
Checked Tools/gencpp submodule, correct version found
Checked Tools/genmsg submodule, correct version found
Checked Tools/jMAVSim submodule, correct version found
Checked Tools/sitl_gazebo submodule, correct version found
Checked cmake/cmake_hexagon submodule, correct version found
Checked mavlink/include/mavlink/v1.0 submodule, correct version found
Checked src/lib/DriverFramework submodule, correct version found
Checked src/lib/dspal submodule, correct version found
Checked src/lib/ecl submodule, correct version found
Checked src/lib/matrix submodule, correct version found
Checked src/modules/uavcan/libuavcan submodule, correct version found
Checked unittests/googletest submodule, correct version found
make -C /c/px4/Firmware/build_px4fmu-v2_default -j4 --no-print-directory
make[1]: *** No targets specified and no makefile found.  Stop.
make: *** [px4fmu-v2_default] Error 2`

ecuzhang, 2016-02-25:
I got the same error either, is the problem of installation or other issues?Thanks for your help.

psiorx, 2016-03-15:
> make[1]: *** No targets specified and no makefile found. Stop.

Try a 
`make clean`
first