PX4/PX4-Autopilot #3711: ERROR: Excessive offset for mag 0

tubeme, 2016-02-09:
Hello,

I cannot find a way to calibrate the compass. Flashed 1.1.3, flashed Stable from QGC, tried QGC 2.9 and 2.9.2. Firstly it gave mi just

[cal] ERROR: Excessive offset for mag 0

After QGC from 2.9 to 2.9.2 it started giving me:

[cal] ERROR: Excessive offset for mag 0
[cal] ERROR: Excessive offset for mag 1
[cal] calibration failed: mag

The Gyro and Accelero calibration is OK. The airspeed calibration is OK so the i2c is working. 
Tried another brand new GPS lea6H the same problem occurs. Boot-up seems normal. Some logs.

[msgs_2016_02_06_16_19_05.txt](https://github.com/PX4/Firmware/files/123326/msgs_2016_02_06_16_19_05.txt)
[msgs_2016_02_06_16_23_04.txt](https://github.com/PX4/Firmware/files/123328/msgs_2016_02_06_16_23_04.txt)
[msgs_2016_02_06_16_42_52.txt](https://github.com/PX4/Firmware/files/123327/msgs_2016_02_06_16_42_52.txt)
[msgs_2016_02_09_15_54_00.txt](https://github.com/PX4/Firmware/files/123325/msgs_2016_02_09_15_54_00.txt)

Any ideas?

LorenzMeier, 2016-02-10:
Not yet, but I'll cross test

LorenzMeier, 2016-02-11:
Do you have a log file from that system so I can look at raw data?

tubeme, 2016-02-11:
Here is data file I have on the SD. There are no regular logs because I cannot arm. I have a bootlog which is boring with PX4IO CRC OK many times...
[dataman.zip](https://github.com/PX4/Firmware/files/127208/dataman.zip)

tubeme, 2016-02-20:
This thing continues in 1.2 beta. I downloaded Today's QGC and flashed Beta from the menu which should be PX4 1.2. Still not able to calbirate the mag... the same reason the same thing.

LorenzMeier, 2016-02-25:
@tubeme I just pushed a change on master which will either fix it for you, or if not, give you in the message panel in QGC an indication of the mag offsets - if it still doesn't work, please send me the numbers you are getting.

tubeme, 2016-02-25:
Calibrated 3 times unsuccessfully moving the telemetry and RC much further away from FC. Thought it might have some interference. 

ERROR: Replace board, fault in mag 0
Offsets:   0.5877,   0.7952,   0.2077

ERROR: Replace board, fault in mag 0
Offsets:   0.5908,   0.7999,   0.2069

ERROR: Replace board, fault in mag 0
Offsets:   0.5841,   0.8030,   0.2027

I suppose Mag0 is the main Mag in our case it is external Quanum Lea6H GPS+Mag.

So I changed it with a new one and again calibrated with the same result.

ERROR: Replace board, fault in mag 0
Offsets:   0.5831,   0.8003,   0.2069

It is similar to the previous calibrations so I wonder if in my case Mag0 is not the internal one

We've had probably the best transitions with this machine with some older master version. Here is the calibration from before (I got it from a Log which I can send if needed): 

CAL_MAG0_ID	73225.0
CAL_MAG0_ROT	0.0
CAL_MAG0_XOFF	1.1486901
CAL_MAG0_XSCALE	0.99373263
CAL_MAG0_YOFF	0.038466614
CAL_MAG0_YSCALE	1.0996851
CAL_MAG0_ZOFF	-0.2766014
CAL_MAG0_ZSCALE	1.0077993
CAL_MAG1_ID	131594.0
CAL_MAG1_ROT	-1.0
CAL_MAG1_XOFF	0.57691956
CAL_MAG1_XSCALE	1.0
CAL_MAG1_YOFF	0.80932355
CAL_MAG1_YSCALE	1.0
CAL_MAG1_ZOFF	0.21076447
CAL_MAG1_ZSCALE	1.0

I notice that the same offsets appear at Mag1 which is the internal Mag.

Here is the startup:

nsh> sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
 1:   SYS_USE_IO: match
 13006: + SYS_AUTOSTART: match
[i] Custom config: /fs/microsd/etc/config.txt
px4io: CRCs match
dataman: Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
bst: no devices found
adc init done
ver hwcmp match: PX4FMU_V2
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
hmc5883: failed to enable sensor calibration mode
hmc5883: calibration failed
hmc5883: no device on bus 1
mpu6000: no device on this bus
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
nsh: sf10a: command not found
px4io default PWM output device
mavlink_if0: mode: 0, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
 157600:   SYS_COMPANION: match
mavlink_if1: mode: 3, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
sdlog2: log buffer size: 12288 bytes
[init] Vehicle type: VTOL
gps: module found: UBX
[i] Mixer: /etc/mixers/vtol_quad_x.main.mix on /dev/pwm_output0
[i] Mixer: /fs/microsd/etc/mixers/vtol_AAERT.aux.mix on /dev/pwm_output1

mavlink_if1: Disabling hardware flow control
ver hwcmp match: PX4FMU_V2
px4flow [217:100]
px4flow: scanning I2C buses for device..
mavlink_if2: mode: 5, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B

Vasil

tubeme, 2016-02-25:
This is the start up with the other GPS+Mag:

nsh> sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
 1:   SYS_USE_IO: match
 13006: + SYS_AUTOSTART: match
[i] Custom config: /fs/microsd/etc/config.txt
px4io: CRCs match
dataman: Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
bst: no devices found
adc init done
ver hwcmp match: PX4FMU_V2
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
hmc5883: ERROR: TIMEOUT 1: Invalid argument
hmc5883: FAILED: SCALE
hmc5883: failed to enable sensor calibration mode
hmc5883: calibration failed
hmc5883: no device on bus 1
mpu6000: no device on this bus
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
Airspeed on I2C bus 1 at 0x28 (bus: 100 KHz, max: 100 KHz)
nsh: sf10a: command not found
px4io default PWM output device
mavlink_if0: mode: 0, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
 157600:   SYS_COMPANION: match
mavlink_if1: mode: 3, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
sdlog2: log buffer size: 12288 bytes
[init] Vehicle type: VTOL
gps: module found: UBX
[i] Mixer: /etc/mixers/vtol_quad_x.main.mix on /dev/pwm_output0
[i] Mixer: /fs/microsd/etc/mixers/vtol_AAERT.aux.mix on /dev/pwm_output1

ver hwcmp match: PX4FMU_V2
px4flow [215:100]
px4flow: scanning I2C buses for device..
mavlink_if2: mode: 5, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B

NuttShell (NSH)
nsh> mavlink_if1: Disabling hardware flow control

LorenzMeier, 2016-02-25:
Its interesting to see that the sensor calibration routine fails for your mag. What is the exact model of it?

LorenzMeier, 2016-02-25:
And if possible share a picture of your wiring. Something is not quite right with your setup.

tubeme, 2016-02-25:
http://www.hobbyking.com/hobbyking/store/__77547__Quanum_LEA_6H_GPS_with_Compass_and_Pedestal_Stand_with_Case_EU_Warehouse_.html?strSearch=quanum%20lea

LorenzMeier, 2016-02-25:
If you disconnect the GPS, can you calibrate successfully?

tubeme, 2016-02-25:
I disconnected only the i2C from the Quanum but still couldn't calibrate. I will try to disconnect the whole GPS.

[image]

LorenzMeier, 2016-02-25:
I'm pretty sure by now that its your internal compass which is failing. Does a different HKPilot32 work?

tubeme, 2016-02-25:
Yes I'm sure it is the internal now. Without the GPS at all still the same error:

ERROR: Replace board, fault in mag 0
Offsets:   0.5960,   0.7991,   0.1934

But I tell you again. With very similar offsets last spring with this master:

Firmware Version	0fed23a0006854ff4936bce2f1b19b3862ccad39

We were flying perfect...

We have some AUAV-X2's so that is what we will try... We have only one HKPilot here...

LorenzMeier, 2016-02-25:
I can prove to you analytically that this mag is not capable of measuring the earth field: The full range is 1.4 Gauss, you have an offset of 0.8 Gauss in Y, the earth field is about 0.6 Gauss large - that means that you will clip measurements in the presence of noise in your Y axis in flight.

The last mag we had like this provided proper measurements after reflow soldering. Replace the board. Its broken. We're not supporting flying with broken hardware.

LorenzMeier, 2016-02-25:
I'm going to add better output calling out if its an internal or external mag, then close this.

tubeme, 2016-02-25:
Even when I connect the Quanum GPS it reports Mag0 fail. IE it does not load properly the external Magnetometer. Which means the there is something more than just Internal Mag failure. Some problem with the Quanum mag. Which probably did nto exist at the time we were calibrating it. We did not fly this machine for quite a while then we used it for bench testing. So we kept the calibration for a lot of time until just 1 month ago when we changed the profile.

tubeme, 2016-02-25:
We will change the board! Thanks

LorenzMeier, 2016-02-25:
I implemented a warning telling you if its the internal or external compass which failed. Closing now.

closed 2016-02-25 by LorenzMeier