PX4/PX4-Autopilot #3713: Kill switch stops PWM outputs, instead of setting a valid minimum pw

kd0aij, 2016-02-10:
The PX4IO lockdown mode is disabling the PWM outputs completely. This is relying on the unspecified behavior of ESCs receiving invalid PWM signals to stop motors. It needs to actively assert a minimum valid pulsewidth on all motor channels to stop them immediately. 

The px4fmu mixing code is OK: it does not disable PWM outputs: motor channels go to 900usec pulsewidth immediately upon assertion of the lockdown flag. So this issue does not affect the PixRacer.

kd0aij, 2016-02-10:
@LorenzMeier Since the "lockdown" function is defined in PX4IO as disabling PWM output (generating no pulses), we should not use it for the kill function when PX4IO is present. 
Could the PX4IO_P_SETUP_ARMING_ALWAYS_PWM_ENABLE flag be set in the kill switch logic as a workaround?

LorenzMeier, 2016-02-12:
I would propose to change lockdown to output the disarmed pulse. This will be either 0 or a safe value.

LorenzMeier, 2016-02-15:
Fix here: https://github.com/PX4/Firmware/pull/3753

kd0aij, 2016-02-16:
Thanks, it would have taken me hours to figure out how that code worked.

closed 2016-02-16 by kd0aij