PX4/PX4-Autopilot #3714: Optical Flow Simulation Broken

jgoppert, 2016-02-10:
#3701 is not functional on master

It seems that inav isn't getting any data. Also QGC isn't working. I did a diff on a regular sim with gazebo_iris and gazebo_iris_opt_flow, and here is the diff

```bash
--- successful	2016-02-10 04:22:17.223721291 -0500
+++ failure	2016-02-10 04:22:24.623721208 -0500
@@ -1,4 +1,6 @@
-init] task name: mainapp
+Built target rotors_gazebo_mavlink_interface
+[init] shell id: 47840548734016
+[init] task name: mainapp
 
 ______  __   __    ___ 
 | ___ \ \ \ / /   /   |
@@ -28,34 +30,35 @@
 [Msg] Waiting for master.
 [Msg] Connected to gazebo master @ http://127.0.0.1:11345
 [Msg] Publicized address: 10.106.48.111
+[Wrn] [gazebo_lidar_plugin.cpp:77] [gazebo_lidar_plugin] Please specify a robotNamespace.
+[Wrn] [gazebo_opticalFlow_plugin.cpp:85] [gazebo_optical_flow_plugin] Please specify a robotNamespace.
 Publishing to: motors
 INFO  Sending initial controls message to simulator
 dataman: Unknown restart, data manager file 'rootfs/fs/microsd/dataman' size is 103090 bytes
+[Wrn] [RaySensor.cc:312] ranges not constructed yet (zero sized)
 INFO  TONE_SET_ALARM 0
 INFO  GYROSIM::init
 INFO  GYROSIM_gyro::init base class ret: 0
-WARN  baro IOCTL 94631
+WARN  baro IOCTL 150523
 Command failed, returned 1
 pwm_out_sim MODE_8PWM
-[Wrn] [Publisher.cc:140] Queue limit reached for topic motors, deleting message. This warning is printed only once.
-[Wrn] [Publisher.cc:140] Queue limit reached for topic motor_speed/0, deleting message. This warning is printed only once.
 [Msg] Connected to gazebo master @ http://127.0.0.1:11345
 [Msg] Publicized address: 10.106.48.111
+[Wrn] [Publisher.cc:140] Queue limit reached for topic motor_speed/0, deleting message. This warning is printed only once.
+[Err] [InsertModelWidget.cc:298] Missing model.config for model "/home/jgoppert/git/Firmware/Tools/sitl_gazebo/models/rotors_description"
+[Wrn] [Publisher.cc:140] Queue limit reached for topic motors, deleting message. This warning is printed only once.
+INFO  simulator is slow. Delay added: 10853 us
 [Wrn] [Publisher.cc:140] Queue limit reached for topic motor_speed/1, deleting message. This warning is printed only once.
 [Wrn] [Publisher.cc:140] Queue limit reached for topic motor_speed/2, deleting message. This warning is printed only once.
-[Err] [InsertModelWidget.cc:298] Missing model.config for model "/home/jgoppert/git/Firmware/Tools/sitl_gazebo/models/rotors_description"
-[Wrn] [Publisher.cc:140] Queue limit reached for topic imu, deleting message. This warning is printed only once.
-[Wrn] [Publisher.cc:140] Queue limit reached for topic motor_speed/3, deleting message. This warning is printed only once.
-[Wrn] [Publisher.cc:140] Queue limit reached for topic gazebo/command/motor_speed, deleting message. This warning is printed only once.
 WARN  pwm_out_sim: No valid fds
-WARN  baro IOCTL 1096073
+WARN  baro IOCTL 1124823
 INFO  GYROSIM::_set_sample_rate 1000003 Hz
 INFO  GYROSIM: Changed sample rate to 500Hz
 INFO  GYROSIM::_set_sample_rate 1000003 Hz
 INFO  GYROSIM: Changed sample rate to 500Hz
-WARN  baro IOCTL 1096373
-WARN  baro IOCTL 1096557
-WARN  baro IOCTL 1096980
+WARN  baro IOCTL 1126289
+WARN  baro IOCTL 1126532
+WARN  baro IOCTL 1127551
 INFO  TONE_SET_ALARM 1
 ERROR px4_task_spawn_cmd: failed to set name of thread 34 0
 
@@ -80,15 +83,14 @@
 WARN  log buffer size: 8192 bytes
 telem> [log] log dir: rootfs/fs/microsd/log/2016-02-10
 WARN  sdlog2: failed setting sched params
-telem> [log] start: 09_20_39.px4log
-pxh> WARN  changing partner IP to: 127.0.0.1
-WARN  pwm_out_sim: No valid fds
-WARN  [inav] GPS signal found
-WARN  init ref: lat=47.3977420, lon=8.5455940, alt=  0.0540
-INFO  home: 47.3977420, 8.5455940, 0.06
-INFO  TONE_SET_ALARM 15
-INFO  TONE_SET_ALARM 0
-INFO  TONE_SET_ALARM 4
-INFO  TONE_SET_ALARM 0
+[Wrn] [Publisher.cc:140] Queue limit reached for topic motor_speed/3, deleting message. This warning is printed only once.
+[Wrn] [Publisher.cc:140] Queue limit reached for topic gazebo/command/motor_speed, deleting message. This warning is printed only once.
+[Wrn] [Publisher.cc:140] Queue limit reached for topic imu, deleting message. This warning is printed only once.
+telem> [log] start: 09_21_40.px4log
+WARN  changing partner IP to: 127.0.0.1
+pxh> WARN  pwm_out_sim: No valid fds
+WARN  INAV poll timeout
+WARN  Q POLL TIMEOUT
+[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.

```

jgoppert, 2016-02-10:
My guess is that the vehicle name is different or something similar. Also, there is a warning about specifying a namespace for the sensors:  

> [Wrn] [gazebo_lidar_plugin.cpp:77] [gazebo_lidar_plugin] Please specify a robotNamespace.
> [Wrn] [gazebo_opticalFlow_plugin.cpp:85] [gazebo_optical_flow_plugin] Please specify a robotNamespace.

ChristophTobler, 2016-02-10:
@jgoppert I just did a new clone of master and the simulation works. I get the two warnings with the robotNamespace but that has no effect on the simulation

jgoppert, 2016-02-10:
Gazebo starts up. No errors.  Just no communication with qgc.

jgoppert, 2016-02-10:
It works for normal gazebo sim, notice how inav works for that build and gets a simulated gps lock, do you get this with opt flow?

jgoppert, 2016-02-10:
Doing a fresh clone myself to confirm.

ChristophTobler, 2016-02-10:
Yes I do get "WARN  [inav] GPS signal found" with the opt flow simulation

jgoppert, 2016-02-10:
The bug is still there for me. I'm running ubuntu 14.04, ros jade, gazebo 1.6.
I'm guessing you are probably running ros indigo?

ChristophTobler, 2016-02-10:
Hm strange... I'm running ubuntu 14.04, ros jade, gazebo 6.5.1. Are you in a network where others are running a simulation as well?

jgoppert, 2016-02-10:
@ChristophTobler no, just using my laptop here. I followed install guidelines for ros with gazebo6 here: http://gazebosim.org/tutorials?tut=ros_wrapper_versions

jgoppert, 2016-02-10:
Using build command: make posix_sitl_default gazebo_iris_opt_flow

ChristophTobler, 2016-02-10:
I use: make posix gazebo_iris_opt_flow. not sure if it makes a difference

ChristophTobler, 2016-02-10:
I just did a normal installation of gazebo I think

jgoppert, 2016-02-10:
Build command didn't make a difference. So instead of apt install you built from source, or installed a binary directly?

jgoppert, 2016-02-10:
I'm running 6.5.1 via apt. Going to try a computer restart.

ChristophTobler, 2016-02-10:
I think I did it like this: http://gazebosim.org/tutorials?tut=install_ubuntu& (Alternative installation: step-by-step). But it still was gazebo 6 back then

ChristophTobler, 2016-02-10:
what qgc version are you running? I'm running 2.9.2

jgoppert, 2016-02-10:
@ChristophTobler I was running it from ppa which as outdated at 2.8.0. Upgrading to 2.9.2 fixed it, thanks!

jgoppert, 2016-02-10:
It also seems to require that I start qgc before the sitl.

closed 2016-02-10 by jgoppert

jgoppert, 2016-02-10:
In fact, I have to shut down qgc and start it again before each run for it to connect, which is kind of weird.

ChristophTobler, 2016-02-10:
I always have to press Disconnect (upper right corner in qgc) before I run a new simulation. Maybe that works for you as well

vinhk, 2016-07-20:
is the issue fix ? I am having trouble as well with making the optical flow display data from the gazebo simulator.