PX4/PX4-Autopilot #3725: integration3: terminal on Snapdragon broken after shutdown

julianoes, 2016-02-11:
When the mainapp is stopped using Ctrl+C on the Snapdragon it leaves the linux shell scrambled.

A workaround for now is to type exit and enter a new shell.

mcharleb, 2016-02-11:
Better workaround is to run:

stty sane

mcharleb, 2016-02-11:
The following code seems to be the culprit:

                // change input mode so that we can manage shell
                struct termios term;
                tcgetattr(0, &term);
                term.c_lflag &= ~ICANON;
                term.c_lflag &= ~ECHO;
                tcsetattr(0, TCSANOW, &term);
                setbuf(stdin, NULL);

Removing it allows terminal to shut down correctly.

LorenzMeier, 2016-02-11:
Its not a solution to kill the arrow-up handling. The solution is to re-instate the previous shell state.

mcharleb, 2016-02-12:
Pushed fix to integration3

julianoes, 2016-02-12:
Tested, works. Thanks @mcharleb.

closed 2016-02-12 by julianoes