PX4/PX4-Autopilot #3726: integration3: rates suddenly slowing down

julianoes, 2016-02-11:
Initially sensors and attitude_estimator_q run with 250 Hz as the MPU publishes at, however, after a seemingly random time, everything slows down and the uorb calls (or polls) seem to end in timeouts.

To debug use the branch https://github.com/PX4/Firmware/tree/integration3_ratedebug.

When it's running fine at the beginning, it should say:
```
[08500/00]  32:08.397  HAP:33:pub sensor combined, interval: 3990  2241  sensors.cpp
```

Then after a while:
```
[08500/00]  34:10.061  HAP:33:pub sensor combined, interval: 181481  2241  sensors.cpp
```

julianoes, 2016-02-11:
My most recent finding is that the measure call from DevObj stop, and thus we get timeouts because the sensors are not publishing anymore.

mcharleb, 2016-02-11:
How can I best reproduce this?

julianoes, 2016-02-11:
@mcharleb The usual
```
(cd /home/linaro && ./mainapp path/to/mainapp.config); exit
```

julianoes, 2016-02-12:
@mcharleb I continued debugging the timeout issue.

Firstly, I think we should be using CLOCK_MONOTONIC in the schedule because CLOCK_REALTIME can skew which is probably not what we want, right?
Can you make sure pthread_cond_timedwait() uses also CLOCK_MONOTONIC internally?
https://github.com/PX4/DriverFramework/commit/ae0c88bee59830bc7896b3c2fdcfd6833ec127af

Also, I found that pthread_cond_timedwait would hang if the time was too small, e.g. < 100 us.
The workaround for now is to do a work item immediately when it's less than 100 us away.
Can you please inspect pthread_cond_timedwait() and see what could be going wrong here?
https://github.com/PX4/DriverFramework/commit/d1b963adef91991fa8b314b307a1651afb8b5a88

The state after the above two fixes was that sensors would run at a constant 4000 us interval (so at the 250 Hz that the gyro driver publishes.
However, the attitude_estimator_q would have intervals around 5000 to 8000 us every now and then (at around 1Hz).
I discovered that the open(mavlink_fd, ...) call raised a couple of px4_ioctl somehow and slowed down the attitude_estimator_q loop down.
I've disabled the open call for now and it lets the attitude_estimator_q run at 250 Hz like sensors.
https://github.com/PX4/Firmware/commit/d82ed390bc45ea6fa463dde730105962d5fd9cd6

Unfortunately, after these changes, there is still trouble, however it's harder to reproduce.
I ran the setup for a longer period of time and saw that both sensors and attitude_estimator_q got timeouts after some time (in one case 18minutes after starting).

From what I can see is that the callback in [DriverFramework.cpp](https://github.com/PX4/DriverFramework/blob/3061277af34daaa4814055bc8da7f3b722d0be3e/framework/src/DriverFramework.cpp#L547) is not called anymore.

I'm digging into this later but it would be helpful if somebody could look at it in parallel.
A debug branch with printfs is pushed to [integration3_ratedebug2](https://github.com/PX4/Firmware/commits/integration3_ratedebug2).

julianoes, 2016-02-12:
FYI @liamstask

mcharleb, 2016-02-13:
@jywilson FYI

julianoes, 2016-02-13:
The timeouts I am seeing are definitely still relating to pthread_cond_timedwait() getting called with an abstime too close. The current workaround of scheduling work closer than 100 us rightaway, gets us to a state where the DriverFramework runs fine for about 30mins but might still suddenly crash at some point. This happens presumably through timing jitter because the sensor read happens **after** getting the current time. I am going to improve the workaround by checking the time again just before calling pthread_cond_timedwait().

I think pthread_cond_timedwait() needs to be fixed internally. If the timeout time has already been reached, it needs to return straightaway instead of block indefinitely.
@mcharleb, @jywilson please comment on this.

Also, I've once been getting a DSP reset saying:
```
[72536.749505] ADSP subsystem failure reason: SFR Init: wdog or kernel error suspected.
```

julianoes, 2016-02-13:
To reproduce this, use the branch [integration3_ratedebug2](https://github.com/PX4/Firmware/tree/integration3_ratedebug2) and let it run for a long time until you see sensors and attitude_estimator_q complaining about long timeouts.

If you don't want to wait too long, play with the [MIN_RESCHEDULE_TIME_US define](https://github.com/PX4/DriverFramework/blob/d2088f44b25ac9fe52a32ff1363930ba26e6f12f/framework/src/DriverFramework.cpp#L54). The lower it's chosen, the more often I expect timeouts.

julianoes, 2016-02-14:
I've added a [2nd workaround](https://github.com/PX4/DriverFramework/commit/d2088f44b25ac9fe52a32ff1363930ba26e6f12f) which makes sure that pthread_cond_timedwait() is never called with a timeout closer than 100us in the future.

Still, after some time, we always get the case where pthread_cond_timedwait() just **never** returns. This happens even with a timeout far away (4000us) in the future.

closed 2016-03-08 by julianoes