PX4/PX4-Autopilot #3727: On-board application list

tuofeichen, 2016-02-11:
Hi @LorenzMeier,
I don't know who's maintaining the pixhawk website, but I would like to suggest we updating this page since some new application has been added, such as the LPE and stuff, and not all of them are  documented.  I think it would be helpful so that people knows what applications are already available and can potentially change things in startup script as needed. 

https://pixhawk.org/firmware/apps/start

P.S. I didn't find relevant information on the dev.px4.io either. But lemme know if I missed anything.

Thank you. 

Best Regards
Francis

julianoes, 2016-02-13:
@tuofeichen new developer information should go into http://dev.px4.io. The old wiki is deprecated.
Feel free to make pull request for it in https://github.com/PX4/Devguide.

LorenzMeier, 2016-05-22:
Outdated.

closed 2016-05-22 by LorenzMeier