PX4/PX4-Autopilot #3728: VTOL: Handle yaw condition correctly

LorenzMeier, 2016-02-11:
@AndreasAntener @sanderux on some waypoints other than takeoff and landing the commanded yaw heading might not be reachable. On some occasions (normal waypoints) we want to weathervane. But on other occasions (aligning before performing a transition) that would be an issue.

After thinking for a bit I would propose to weather-vane for all waypoints expect for a waypoint when followed by a transition. Failure to align should result in a transition abort there.

What do you think?

AndreasAntener, 2016-02-11:
That sounds like a good approach.

sanderux, 2016-02-11:
Perhaps we should add an acceptance angle to a waypoint where a transition forward follows

sanderux, 2016-02-11:
and on all the other waypoints a timeout on _yaw_reached, often the MC forward flight will eventually pull the airframe in the desired yaw so i see no problem with continued attempts to correct yaw.

And maybe as a rule (default configs) the yaw rate should be lower on quadplanes

LorenzMeier, 2016-02-11:
You're right. We probably should start with just a timeout for VTOL and then accept if position is right.

sanderux, 2016-02-11:
Any suggested default value for VTOL_WV_WP_TMT (in seconds)?
And I propose that -1 be disabled and 0 considered instant WV.

If we are in agreement i'll get working on this.

AndreasAntener, 2016-02-11:
I think we're taking about a different timeout. Not to weather vane, but to accept the mission waypoint.

sanderux, 2016-02-11:
sorry, thats what i meant, i call it weathervane as it would in effect weathervane when traveling to the next waypoint. But that might be confusing to the user.

So, VTOL_WP_YAW_TMT ?

sanderux, 2016-02-11:
Or maybe add this function globally and disable it by default except on VTOL's?

AndreasAntener, 2016-02-11:
Maybe a mission parameter MIS_YAW_TMT, default -1 and only the VTOL config will have it set to X.

AndreasAntener, 2016-02-13:
@sanderux @LorenzMeier Do we agree that we don't care about heading on the waypoints in general for VTOL? So I'd like to change my timeout impl to work as follows, maybe you always thought about it this way (I didn't):

- default, -1, means we wait on yaw (multicopter)
- >= 0 we don't care for yaw on waypoints, except if we have one where we need to force it, then we consider the timeout

How does that sound? My current implementation uses the timeout for all waypoints which makes it hard to set the value right: small enough so it doesn't block the mission too much but long enough so the transition aligning has enough time.

sanderux, 2016-02-14:
I agree that heading isnt that important, mostly because the heading would align by flying towards the next waypoint (when there is not too much wind).

I would not set a timeout on yaw before transition, that could result in dangerous situations. If we want to support weathervaning on takeoff we need to make this VERY clear.

We now have WV_LTR and WV_LND. I tested LTR and that works fine. I would propose adding WV_NAV and WV_TKF.

We could use the same logic and in case of NAV simply ignore the yaw_reached for normal waypoints. In case of TKF even ignore it on forward transition.

AndreasAntener, 2016-02-14:
> I would not set a timeout on yaw before transition, that could result in dangerous situations. If we want to support weathervaning on takeoff we need to make this VERY clear.

Maybe I wasn't clear, if the timeout occurs, it would abort the mission at that point.

sanderux, 2016-02-14:
We could also implement a 2 stage transition if yaw is unreachable. In stage one it engages the pusher to fly towards the next wp only to align yaw with use of the rudder, in stage 2 it attempts transition

AndreasAntener, 2016-02-14:
Ok I'll finish the timeout as I proposed because it's the simplest way to unblock the current issue.

Any additional yaw handling that you're suggesting is actually independent. Let's say we implement this strategy, then the timeout could be increased but it would still protect it from not reaching the target heading before a transition.

Also, not caring for yaw in general is complementary and necessary to WV on normal waypoints because there it would never reach yaw. So this makes perfect sense if we put everything together after.

AndreasAntener, 2016-02-14:
Implementation updated and SITL tested.

sanderux, 2016-02-14:
https://github.com/PX4/Firmware/pull/3740

PR #3740 (2016-02-16): Yaw timeout for VTOL forwards transition
This adds a yaw timeout on accepting the waypoint for the VTOL forwards transition because a VTOL might not reach yaw in wind. Fixes #3728

MIS_YAW_TMT
- default: -1
- > 0: timeout active on waypoints which force heading, ignore heading for acceptance on normal waypoints

MIS_YAW_TMT set to 10 seconds for default VTOL configurations
closed 2016-02-16 by AndreasAntener