PX4/PX4-Autopilot #3729: RTL returned to previous location

AndreasAntener, 2016-02-11:
After transition failsafe triggered RTL it came back to home but instead of landing it went off again into the direction it came from.

I had RTL_LAND_DELAY set to 0 so it should have landed at home.

http://logs.uaventure.com/view/jWxYmtjVJUSsdRcp27Damm

@sanderux FYI. didn't do a SITL test yet.

sanderux, 2016-02-11:
Maybe this has to do with the new work_item approach, it was built on the previous method. I'll have a look

sanderux, 2016-02-11:
I have successfully reproduced in SITL and identified the problem
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission_block.cpp#L424-L433
(transition to MC before landing)

Now need to find a solution

AndreasAntener, 2016-02-11:
Ah of course.. it resets the transition failsafe state.

I had to add the reset here https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/vtol_att_control_main.cpp#L444 so you can start a transition again in the same flight.

Simplest solution right now would be to only issue the transition command if is_vtol && !is_rotary_wing

sanderux, 2016-02-11:
yes, i just found that.
could we not better reset the transition failsafe only when out flight mode no longer equals RTL?

AndreasAntener, 2016-02-11:
The problem with that is the flight mode actually never changes. I think just issuing the transition command when actually necessary is the most straight forward change and clearly understandable.

AndreasAntener, 2016-02-11:
Revising my comment above, the navigation state changes of course. Still.

sanderux, 2016-02-11:
Well, i was thinking about the behavior when it actually is fixedwing transitioning back, would it not have the same problem in that case? well currently i think transition failure is the only failure that enters RTL so in practice it would help. we do need to pay attention as i have more future RTL failures in mind (like unfeasable distance at certain wind speed)

sanderux, 2016-02-11:
on the other hand, only transitioning when actually fixed wing is ofcourse a good idea anyway. i'll push a fix

AndreasAntener, 2016-02-11:
Your right, FW back transition would have the same problem. But we currently can't get into this situation. In case we will have a back transition failsafe at some point, this would become very interesting since it probably wouldn't do any good anymore to issue a back transition command in RTL :).

sanderux, 2016-02-11:
The scenario i consider is;
VTOL takes off and transitions, then takes measurements of airspeed vs groundspeed, based on current draw and available power it comes to the collusion the mission is not feasible. RTL kicks in and will fly as fixed wing back home, transition will occur but at that point it would start the mission again and loop.

AndreasAntener, 2016-02-11:
Ok but that's different. This would be a separate failsafe a presume. The back transition in RTL only resets the transition failsafe.

sanderux, 2016-02-11:
ah, yes, thats where you are right again :)
Ik, fix is in, i'll push

closed 2016-02-11 by AndreasAntener