PX4/PX4-Autopilot #3732: TECS altitude hold

AndreasAntener, 2016-02-12:
My favourite topic: how does TECS hold altitude. Flown with #3678

I'm trying to analyze two instances:

[image]
http://logs.uaventure.com/view/zrRkhmi35oGRRZiq9fTcPm

Why is the altitude setpoint resetting here twice, even though I'm still pitching up?

[image]
http://logs.uaventure.com/view/Gi8kX8ho4oaye8gcjjS5sd

As far as I understand, after I let go of pitch the altitude setpoint is going to align itself with current altitude which makes sense, but then it goes down even further. Is this because of the airspeed difference? (If that assumption is correct it might be prevented by setting a different speed weight)

LorenzMeier, 2016-02-18:
Still unclear, or fixed today?

AndreasAntener, 2016-02-19:
I'll analyze those instances again (because I was looking at the wrong speed) and also flights from yesterday with different speed weight.

AndreasAntener, 2016-02-20:
In our flights TAS and IAS are always the same so my questions from above remain. Generally I was happy with altitude tracking in the latest mission flights. I will need to do a few alt hold flights with priority on height to compare.

AndreasAntener, 2016-02-24:
Ok, the initial setpoint is governed by the FW pos controller: https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1136
Then TECS controls the rate at which its setpoint is going to converge with the initial setpoint.

Problem A:
The initial setpoint is set at the time when pitch input is in side the dead zone again. This means as soon as the user lets go of pitch we will start trying to hold altitude again, that's fine. But which altitude should we try to hold I'm wondering. Currently it is the altitude at that very moment. During leveling out altitude still changes which it then has to correct again. As a pilot I expect the plane to finish the climb/descent when I let go of the sticks which means it would be a bit higher/lower after leveling out. That's the point in time when the altitude setpoint should be set.

So the final setpoint should look like in the following plot (on the bottom, thick red line instead of thin yellow line):
[image]

Maybe this could be achieved by setting a 0 height rate demand in TECS until the energy error is near 0 and then take current altitude.

Problem B:
In addition, the TECS height rate demand over/undershoots, I guess I could try different height P/FF parameters for this.

closed 2016-11-19 by LorenzMeier