PX4/PX4-Autopilot #3733: Submodules: don't reset on make

julianoes, 2016-02-12:
Git reseting the submodule on make is just not intuitive.

This would be what gets me every time:

1. I add and commit something in a submodule.
2. Now before I want to push it from the submodule and before commiting it in Firmware, I like to compile and test it one more time to make sure I don't push any rubbish.
3. I do make and run it.
4. I realize the bug I was trying to fix in the first place and start debugging again.
5. At some point I realize that my fix has gone from the source :facepunch: 

Arrgh, I'm a developer, I know git, and I don't want ```make``` messing with it!

davids5, 2016-02-12:
Will setting [GIT_SUBMODULES_ARE_EVIL] (https://github.com/PX4/Firmware/blob/master/Tools/check_submodules.sh#L3)
Solve your problem?

LorenzMeier, 2016-02-12:
It must be a corner case of the version check script. Its not intended to work that way and it does work great for the cases where you are missing commits by asking if you would like to update.

julianoes, 2016-02-23:
This just got me again.

I did: 

```
make clean
cd src/lib/DriverFramework
git checkout some_pull_request_branch
cd -
make
```

And the checked out commit in src/lib/DriverFramework is gone.

LorenzMeier, 2016-02-23:
The issue is the make clean step. If you do this the system has no way of knowing wether you did a fresh clone (requiring init) or not. It would need to keep its own registry which would never reset by anything the user does, which is also a risk.

I'm on the fence of calling this an usage issue just requiring a printout to remind you to not do this. Because if we don't wipe out all files on make clean, we get ourselves into the stale file problem.

julianoes, 2016-02-23:
I don't understand, I'm doing my changes **after** the make clean.

LorenzMeier, 2016-02-23:
Yes, but your directory looks the same as a freshly cloned one.

julianoes, 2016-02-23:
Yes, but the submodules are already there, so no need to mess with them.

LorenzMeier, 2016-02-23:
How about you write a script checking their init status? :P.

julianoes, 2016-02-23:
Sure, I'm just wondering where the current check happens.

Is it based on this https://github.com/PX4/Firmware/blob/master/Tools/check_submodules.sh#L9 ?

LorenzMeier, 2016-02-23:
No, the one biting you is in the Makefile:
https://github.com/PX4/Firmware/blob/d2041b0ef3ffec1fc606c1389b1b74f1cf51beb7/Makefile#L112

julianoes, 2016-02-23:
I was able to fix my issue using check_submodules. I don't know what the line in Makefile does.

julianoes, 2016-02-23:
I removed that line, so now I expect it to work.

julianoes, 2016-04-27:
This has been fixed, closing.

commit c5b1e791 (2016-02-12): Update git check script. Fixes #3733
closed 2016-04-27 by julianoes