PX4/PX4-Autopilot #3739: Problems before or after arming

AndreasAntener, 2016-02-12:
Since a while we have issues with arming or before arming. I'd say 1 out of 5 boots, maybe less.

There are 2 symptoms:
- IO_ARM_DENIED (see px4io status below)
- it arms but props don't spin up: http://logs.uaventure.com/view/u4vCAaB9uiza9ALDetymKn

I observed this on our Fun Cub and on our F450

I know it's not much to go for right now. Maybe we can try to record some additional info if this happens again?

```
px4io: loaded
protocol 4 hardware 2 bootloader 3 buffer 64B crc 0x1c579c6e
8 controls 8 actuators 18 R/C inputs 2 analog inputs 0 relays
696 bytes free
status 0x2765 OUTPUTS_ARMED SAFETY_SAFE RC_OK SBUS FMU_OK MIXER_OK ARM_SYNC INIT_OK
alarms 0x0010 FMU_LOST
vservo 5283 mV vservo scale 10000
vrssi 1651
actuators -10000 -10000 -10000 -10000 0 0 0 0
servos 900 900 900 900 0 0 0 0
reversed outputs: [________] trims: r:  -0.0256 p:   0.0000 y:  -0.3068
18 raw R/C inputs 1515 1515 1094 1514 1094 1094 1094 1094 1094 1094 1094 1094 1515 1515 1515 1515 1998 2998
R/C flags: 0x0018 MAPPING_OK
mapped R/C inputs 0x002f 0:0 1:0 2:0 3:0 5:-8120
ADC inputs 2197 2060
features 0x0008 RSSI_ADC
arming 0x0430 FMU_DISARMED IO_ARM_DENIED INAIR_RESTART_OK ALWAYS_PWM_ENABLE OVERRIDE_IMMEDIATE
rates 0x000f default 50 alt 400
debuglevel 0
controls 0: 2322 659 746 0 0 0 0 0
controls 1: 0 0 0 0 0 0 0 0
controls 2: 0 0 0 0 0 0 0 0
controls 3: 0 0 0 0 0 -8120 0 0
input 0 min 1094 center 1516 max 1934 deadzone 10 assigned 0 options 0x0001 ENABLED
input 1 min 1094 center 1513 max 1934 deadzone 10 assigned 1 options 0x0003 ENABLED REVERSED
input 2 min 1094 center 1094 max 1934 deadzone 10 assigned 3 options 0x0001 ENABLED
input 3 min 1094 center 1513 max 1934 deadzone 10 assigned 2 options 0x0001 ENABLED
input 4 min 1000 center 1500 max 2000 deadzone 10 assigned 100 options 0x0001 ENABLED
input 5 min 1094 center 1514 max 1934 deadzone 10 assigned 255 options 0x0000
input 6 min 1000 center 1500 max 2000 deadzone 10 assigned 255 options 0x0000
input 7 min 1094 center 1514 max 1934 deadzone 10 assigned 255 options 0x0000
input 8 min 1000 center 1500 max 2000 deadzone 0 assigned 5 options 0x0001 ENABLED
input 9 min 1094 center 1514 max 1934 deadzone 0 assigned 255 options 0x0000
input 10 min 1094 center 1514 max 1934 deadzone 0 assigned 255 options 0x0000
input 11 min 1094 center 1514 max 1934 deadzone 0 assigned 255 options 0x0000
input 12 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 13 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 14 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 15 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 16 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 17 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
failsafe 900 900 900 900 0 0 0 0
disarmed values 900 900 900 900 0 0 0 0
nsh> 
 ```

AndreasAntener, 2016-02-12:
Pre and postflight log of the mentioned log above: https://gist.github.com/AndreasAntener/f171c308466e26a55d01

AndreasAntener, 2016-02-12:
Is there something wrong with the IO latency counter?
```
io latency: 2132 events, 0 overruns, 122674937284us elapsed, 57539839us avg, min 53266284us max 61814520us 32397274.000us rms
```

LorenzMeier, 2016-02-12:
Looks very fishy.

LorenzMeier, 2016-02-12:
Its so unrealistically wrong that I rather think memory error, not counter error.

LorenzMeier, 2016-02-12:
Conclusion: Not optimal for VTOL. The counter setup does not work for VTOL yet.

LorenzMeier, 2016-02-13:
The log shows no airspeed - did the system boot fully in this instance? It doesn't look like it:
http://logs.uaventure.com/view/u4vCAaB9uiza9ALDetymKn

Still looking for the root cause.

LorenzMeier, 2016-02-13:
@AndreasAntener Still digging into this. Just to let you know it's not been forgotten.

AndreasAntener, 2016-02-13:
Thanks :) If there's anything we can do to try narrow it down on the field when it happens let me know.

AndreasAntener, 2016-02-24:
Hit this again today:
https://gist.github.com/AndreasAntener/ab5db36c67bf0dcb6d2e
Have a look at the actuator outputs: http://logs.uaventure.com/view/2SN4iEjw6XSKHwnCHwv2RR#Actuators_PLOT

AndreasAntener, 2016-02-24:
Note: the log is from our F450 where I have it set to start logging at boot.

LorenzMeier, 2016-05-22:
We've fixed a couple of issues there in the meanwhile, closing. Please reopen if it happens again.

closed 2016-05-22 by LorenzMeier