PX4/PX4-Autopilot #3741: Transition failsafe for Tailsitter

AndreasAntener, 2016-02-12:
2 todos:
- implement front transition timeout
- check if aborting the forwards transition works properly (instantaneous switch to MC when already pitching 90deg forwards)

@tumbili @sanderux aborting the forwards transition in SITL resulted in the tailsitter crashing for Sanders.

tumbili, 2016-02-12:
@AndreasAntener You mean in simulation, right?

AndreasAntener, 2016-02-12:
Yes

AndreasAntener, 2016-02-14:
@tumbili could you verify if the tailsitter has an actual problem when aborting the forwards transition? I'd like to move this issue out of the robustness milestone if there is no bug.

AndreasAntener, 2016-02-19:
@tumbili Roman I think you tested yesterday that the instant back transition works?

tumbili, 2016-02-19:
@AndreasAntener Yes, that seemed to work, tested it a few times, no issues.

AndreasAntener, 2016-02-19:
Great thank, I updated the description

sanderux, 2016-02-19:
@tumbili did you test back trans during the P1 stage which is quite short?

AndreasAntener, 2016-03-01:
@sanderux do you want to have another look at transition failsafe for tailsitter and tiltrotor? I believe Roman fixed the existing tailsitter issues.

LorenzMeier, 2016-07-29:
Where is this standing?

closed 2018-02-21 by PX4BuildBot