PX4/PX4-Autopilot #3742: code cleanup: sensors orientation mess

petekol, 2016-02-12:
sensors gyro/accel/mag code does the following:
"to align the sensor axes with the board, x and y need to be flipped and y needs to be negated"
even thought there is a rotation step for every sensor

it is highly confusing and not documentation friendly

LorenzMeier, 2016-02-13:
There is not a per-sensor rotation implementation for all drivers. It would be very welcome to have a general API for this and a pull request tested on hardware and implementing it for all sensors would be welcome. You can open a pull request to for the further discussion.

closed 2016-02-13 by LorenzMeier

petekol, 2016-02-15:
new API is good but years to take...
what can be done now is to replace code in hmc5883/MPU6000/MPU925:

"swap axies + rotate X" with rotate Y