PX4/PX4-Autopilot #3746: Airspeed I2C failure, possible error detection?

AndreasAntener, 2016-02-13:
This is from an incident with airspeed on a VTOL flight where the airspeed gave erratic readings:

log: http://logs.uaventure.com/view/a8bSFnXWsFTf57zvAAu2oV
preflight: https://gist.github.com/AndreasAntener/715abd0dcafe91299bca

The problem was a damaged cable. The preflight log shows a lot of comms errors and overflows for airspeed. Could we improve error detection?

AndreasAntener, 2016-02-13:
The readings it gave also clearly point to a problem if you look at the plot, so maybe a more general detection on that level could be done as well.

tubeme, 2016-02-13:
We have more than 1000 hours on PX4 now. We've had 27 crashes because of Airspeed sensor problems in our VTOL configuration. This problems go back before the summer 2015 with really nasty erratic airspeed sensor behavior with some master builds. Now we stick with 1.1.0 and we are trying 1.1.3 with good results so far. You have to make sure the i2c is shielded as well as the airspeed brick is no closer to battery or other RF emitting device. We've had similar irrational readings when the brick was close to the battery. We've had so much problems with Airspeed, so now we included in our pre-flight check a visual inspection of the airspeed reading in QGroundControl prior any flight.

LorenzMeier, 2016-02-13:
You will want to feed the airspeed data to an instance of the data validator:
https://github.com/PX4/ecl/blob/0a9f7e58dbb5708fc894a865376ede6cbb675fd4/validation/data_validator.h

It does watch the error count and variance. Really low variance, really extreme variance and a high error count are all reasons to trigger. Works fine for gyro / accel / mag data.

SimonWilks, 2016-02-14:
@tubeme would you have logs from some of the more recent failures that we could look at? It would also be interesting to categorise the failures you experienced including how often any one of those failure types occurred. Of the failures you experienced, what proportion could have been detected prior to takeoff with a pre-flight check vs ones that only occur in flight.

sanderux, 2016-02-14:
In addition to checking for eronous readings, perhaps crashes can be avoided by always engaging MC when airspeed reads below X, and having a minimum time that airspeed must read above x before disabling MC.

sanderux, 2016-02-14:
Also, i noticed that my erronous readings went away when resetting (choosing a new airframe). I believe there is a problem in the calib procedure when there is already a value.

LorenzMeier, 2016-02-14:
@sanderux It also depends on the exact supply voltage - you need to get one of these level shifters and mount it. Its far more likely that's the root cause:

https://www.sparkfun.com/products/11955

sanderux, 2016-02-14:
@LorenzMeier Looking at http://logs.uaventure.com/view/j8WNK662oewGxao32TXqBF#A_PitchRate_PLOT (indoor hover)
i see 2 spikes. far less then the eratic behavior from before. my PM was malfunctioning, replaced that

LorenzMeier, 2016-02-14:
You still need the level shifter to be safe. I will look into the airspeed monitor extension. Need to see where it fits in the architecture.

sanderux, 2016-02-14:
I think that the time between system start and switching to mission would be enough to detect erroneous readings.  this could trigger a sensor failsafe and refuse mission mode.

tubeme, 2016-02-15:
Here are some crashes since NOV 2015:

Plots: http://logs.uaventure.com/view/o945UKvhfqLWtGDC5Muu63
Log: http://logs.uaventure.com/logmuncher/charts/o945UKvhfqLWtGDC5Muu63/logmuncher.log

Plots: http://logs.uaventure.com/view/qQTRD6d8KJpKYtfTRbFCND
Log: http://logs.uaventure.com/logmuncher/charts/qQTRD6d8KJpKYtfTRbFCND/logmuncher.log

Plots: http://logs.uaventure.com/view/e3wnvFx9FnWwJghKjn7e4B
Log: http://logs.uaventure.com/logmuncher/charts/e3wnvFx9FnWwJghKjn7e4B/logmuncher.log

tubeme, 2016-02-15:
@SimonWilks The rule for us now is that if the airspeed starts correct and it is not showing erratic behavior in the QGroundControl that most likely it will behave correctly during the whole flight. Now it is even easier to see in QGC because since 2.8 when aircraft is on ground the QGC pane shows the ground speed. If the airpseed did not initialize correctly for some reason the AS will start jumping up and down and show on the QGC widget. Usually after restart, or couple of restarts things are OK. Sometimes Calibration does the trick. We've had days with 2-3 Airspeed sensor calibrations in between the flights... We couldn't find a pattern of behavior. One day the airspeed sensor works rock solid. The next morning it just ditches the plane in the transition or in flight... After the AS pixracer voltage argument got on the groups I started measuring the voltage the Pixhawk gives to the AS sensor. For some reason on every restart I got a little different voltage reading from the Pixhawk. On couple of occupations i got 4.6 volts from the i2c dont know if it has a connection... Strange things happen and we never fly without GCS which is absurd. Sometimes we just want to go the field and start flying. We have to be more couscous now. This sensor is not a reliable solution, given that the whole PX4 depend heavily on the AS sensor. I think that now is the moment to drastically change the design. Not just minor cosmetic changes because of the 3v vs 5v problem. The company that sells the AS chip has tens of other variants of AS sensors sure it might be a little more expensive but it is a must in such an important vital part of the autopilot. Everybody is willing to pay the extra cost for reliability and ease of use and DJI prove it. We ordered the EagleTree sensor to see if it works better...

sanderux, 2016-02-15:
As per Lorenz's advice, i have a couple of these underway;
https://www.sparkfun.com/products/11955
It will take up to 4 weeks to arrive but once i get them i'll start testing.

Nevertheless i think it is possible for the autopilot to cope with a failing AS;
- Monitor altitude
- Set minimum throttle in auto mode
- For VTOL transitions set a minimum time for transitions
- For VTOL during transition set the SP altitude a little higher and monitor if the altitude has reached the SP, thereby deducing that the FW has provided lift,
- Require multiple sequential readings higher then threshold to consider a threshold met.

LorenzMeier, 2016-02-18:
Done in master. Might still need work, but more on the probe shape and I2C bus HW side.

closed 2016-02-18 by LorenzMeier