PX4/PX4-Autopilot #3747: Secure Mavlink

jimdsouza, 2016-02-13:
I've been using this firmware on my Pixhawk and noticed that anyone with a gcs can access my UAV.

Is there a feature planned to implement authentication (with/without) encryption on the Mavlink between the drone and the GCS?

LorenzMeier, 2016-02-13:
Yes, there is a MAVLink 2.0 testing version which adds authentication. Would you be able to lend a hand with the development work needed to validate it?

jimdsouza, 2016-02-13:
Sure. I was going to try to do just that on a fork of the firmware. But working with the main team on it is better than that. How do I start?

LorenzMeier, 2016-02-13:
Fork the repository on Github, create a new branch off master, update MAVLink, then send a pull request for review & integration.

MAVLink 2.0 update:
http://discuss.dronecode.org/t/mavlink-2-0-progress-tracking/23/4

Associated GCS issue:
https://github.com/mavlink/qgroundcontrol/issues/2800

jimdsouza, 2016-02-13:
Okay, thanks!

jimdsouza, 2016-02-13:
Sorry, I'm new to this project and github. Just to reiterate, I fork PX4/Firmware, and I fork diydrones/mavlink. I then branch the former and use the mavlink2-wip branch in diydrones/mavlink to update the branch of PX4/Firmware. I then test the code and send in a pull request to PX4/Firmware.

Is that correct?

LorenzMeier, 2016-02-13:
You generate the C-library for MAVLink using the mavlink fork and put the files in Firmware/mavlink/... in the right location, then you build it.

jimdsouza, 2016-02-15:
I built it. Should I send a pull request?

LorenzMeier, 2016-02-15:
Yes please! Can you upload a ZIP file with the generated MAVLink files somewhere? Then Don can have a go in QGroundControl as well. @DonLakeFlyer FYI

jimdsouza, 2016-02-15:
Sure! You can download it from [here](http://jimdsouza.com/mavlink2.zip).

LorenzMeier, 2016-05-22:
@jimdsouza MAVLink 2.0 is now on master. If you would like to implement signing a PR would be appreciated.

closed 2016-05-22 by LorenzMeier

jimdsouza, 2016-05-22:
Thanks, I'll try it. Not sure what you mean by signing a PR.

LorenzMeier, 2016-05-22:
We haven't enabled signing of messages yet. If you implement that, please send a pull request.

jimdsouza, 2016-05-22:
Sure, I'll give it go.