PX4/PX4-Autopilot #3760: Follow-Me: Implement flight mode support (AUTO: Follow Me) and navigator support

LorenzMeier, 2016-02-15:
@catch-twenty-two Here is where you would need to dive into:

You need to create a new navigator mode:
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/land.cpp

And register it here:
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L485

Add a new message handler for the new MAVLInk message:
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L203

And add the logic to plumb the new flight mode through MAVLink so QGC can see it:
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L161

And finally enable the new mode in the commander:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp#L365
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L642

If you edit all these locations and fix all resulting compile errors you should be good to have the new flight mode entirely in the system. You then can implement everything in your new file in the navigator.

catch-twenty-two, 2016-02-16:
@LorenzMeier Thanks!

LorenzMeier, 2016-03-28:
Done by @catch-twenty-two, thanks!

closed 2016-03-28 by LorenzMeier

VTOLDavid, 2018-11-02:
Is it possible to activate this mode from mission waypoint?