PX4/PX4-Autopilot #3775: No RC control when in failsafe

sanderux, 2016-02-16:
Here's a log from a (vtol) flight that went into failsafe due to low  battery voltage (it was set a bit too high).
http://logs.uaventure.com/view/3msrKYiSjKrDvKKckveA4e

The as there was no autoland engaged the plane went into loiter.
I was unable to retake control of the plane by switching the mode switch on the RC, i had to switch the mode on the GCS to regain control

sanderux, 2016-02-16:
RX issue due to TX signal loss. Devention RX will not allow mode switching after switching to RTL

closed 2016-02-16 by sanderux

kd0aij, 2016-02-16:
would this have been a fly-away if no GCS were connected?

sanderux, 2016-02-16:
yes, well, it would loiter until empty bat. but it's my RX, that sets channel 5 to 2000 at signal loss but doesnt recover when reconnected. I will replace the RX