PX4/PX4-Autopilot #3776: VTOL: Implement moving using fw motor

tumbili, 2016-02-16:
In the winds the VTOL now tries to move forward by pitching down. This creates negative lift when the plane is facing into the wind and the thrust setpoint goes up. We've encountered cases where the position controller would disable horizontal position control in order not to decrease altitude.
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1635
The position controller would then get into a cycle of enabling and disabling horizontal position control which leads to large oscillations of the pitch setpoint.
Log: http://logs.uaventure.com/view/W8rC6nV8y5C9dmfMYoxQsd

tumbili, 2016-02-16:
This can be fixed by allowing the MC controller to move around using the fw motor instead of using pitch / roll.

sanderux, 2016-02-16:
implementing this will also benefit yaw behavior

sanderux, 2016-02-24:
first draft, works in sitl:https://github.com/PX4/Firmware/pull/3838

LorenzMeier, 2016-02-24:
What you did there will de-stabilize the attitude control. What we will likely need instead is to split attitude and lateral acceleration demand in the position controller (sort of the direction of the trust vector vs. the direction of the desired attitude) and the attitude controller then either combines them for an under actuated vehicle (like a quad) or looks at them separately for a fully actuated or over actuated vehicle. @tumbili Please pitch in.

The interesting point about that concept is that it will scale to tilt rotors or anything like that and will naturally give you the right tilt angle + the right rotor speeds for what you want to achieve.

sanderux, 2016-02-24:
I know, it's just a proof of concept now. 
I'm actually amazed at how well it's holding attitude in SITL now.
We could include setpoints to determine lateral acceleration but i think we could also include airspeed to detect if we are flying in nose wind. @tumbili what do you think?

sanderux, 2016-02-24:
/cc @AndreasAntener

closed 2016-04-04 by tumbili