PX4/PX4-Autopilot #3778: Travis CI for OSX 5x slower than Ubuntu

jgoppert, 2016-02-16:
Do we know why this is happening? It is a pain to have to wait 30 + minutes for CI. If we could figure out what is wrong and get it on par with ubuntu it would speed up development quite a bit.

dagar, 2016-02-16:
The answer is probably just "it's slow", but I'll take a look. One option is having the build do much less on OSX. Do the nuttx builds ever break on one platform, but not the other? Maybe OSX only needs to be building posix.

LorenzMeier, 2016-02-16:
Yes, POSIX only would be enough on OS X. That would allow us to skip the arm compiler install, which should speed it up dramatically. Please only comment out, not remove those bits though, thanks!

DonLakeFlyer, 2016-02-16:
Same thing with QGC OSX builds, make me crazy!

PR #3791 (2016-02-18): travis-ci only build sitl on OSX
-fixes #3778
closed 2016-02-18 by LorenzMeier