PX4/PX4-Autopilot #3784: Helicopter: Strange Behaviour

tubeme, 2016-02-17:
Hello guys,

We are building IC classic hellicopter with Pixhawk. 

[image]

So here is the first bench test we made. For the bench test we strapped the Heli to an Euro Pallet to be sure it won't do some silly thing. It is pretty dangerous machine for me, because 1. blades are pretty big 2. it is with internal combustion so much bigger danger factor.

We start the engine in Acro. We couldn't arm in Acro (I saw this in the Racer thread), but we arm in Manual and then switch to Acro. On Acro the Helicpter was performing as expected. Here is the mixer file we are using:

[coax.main.mix.zip](https://github.com/PX4/Firmware/files/133938/coax.main.mix.zip)

We are using the Coax profile with our custom mixer.

In the experiment we wanted to see how the Heli will perform on Manual ie. Stabalized. It is done by getting to hover throttle (and the common pitch of course) and then switch to Manual.

So we switched between Acro and Manual and strange things happened.

Fist try and the heli rolled heavily and pitched backwords.

On the second try we held Manual for longer. This time it only pitched backwords. 

It is strange and I cannot see setpoints in the Log file. The Helicopter is perfectly leveled (measured) with the Euro pallet. I expected when switched to manual just to make a small corrections here and there but still perform as in Acro given the leveled position of the craft.

If this was in the air both times it was 100% crash. We don't want to risk it so we want to clear the issue with your help before proceeding in the Air.

Here is the Log:

http://logs.uaventure.com/view/TUABTsVhg9JaZKK76mRyz4

Any ideas help?

kd0aij, 2016-02-17:
testing with the heli tied down is inviting problems with resonance. I once
started a large table walking across the shop floor that way.
If that wasn't the case for you this time, beware that ground resonance can
very quickly destroy your heli also. Perhaps your harness provides enough
damping to avoid the problem though.
I'd suggest starting with low PID gains and slowly increasing them till
you've demonstrated proper control response. Then lose the tie downs.

tubeme, 2016-02-17:
@kd0aij We have this in mind. The Heli constructor and builder has 30 years of experience. Also in Acro the Heli performed as expected at hover throttle and at higher throttle. The problems start with Manual. The PID's are on the safe side, it is obvious from the Log. When this happened I noticed on the horizon indicator it seemed like the heli was going full pitch up ie the indicator went in the green. Despite it was staying flat horizontal. The only thing that moved was the rotor blade.

kd0aij, 2016-02-17:
It seems clear from your log that vibration is saturating your accelerometers and causing the IMU to fail. What kind of vibration isolation are you using?

tubeme, 2016-02-18:
@kd0aij Thanks a lot! We stay connected to GCS during the whole bench test and I'm observing it. The dampening is not so good so we have to work on it more obviously. Heli's are typical with their vibration, especially 2 bladed helis like our model. Thanks again.

Jomhuang, 2016-02-23:
In Acro mode, your RC's input is anglerate setpoint, there is just one controller(rate controller) in pixhawk, so no attitude setpoint. In Manual mode , your RC's input is attitude setpoint.

rousselmanu, 2016-02-23:
@tubeme I'm flying with a small coaxial helicopter and observed the same behaviour on accelerometer measurements (especially on the Z axis which takes a strong offset and measures 0 or so instead of -1 g). It was improved by careful isolation of the rotors: the pixhawk is fixed to the heavy and non-moving parts (batteries, structure...) while the rotor and moving parts are loosely coupled using vibration isolators. This gives better results than isolating the pixhawk alone. Or even better results where obtained by using a higher-end external IMU connected to the pixhawk (cf https://github.com/PX4/Firmware/pull/3123)...

tubeme, 2016-02-23:
@Jomhuang That was exactly my point. In the condition that the Helicopter is perfectly leveled and strapped there should not be any difference between the angle-rate setpoint and the attitude setpoint in the two modes when I switch between them.  As @kd0aij pointed we have a saturation of the accelerator.

tubeme, 2016-02-23:
@rousselmanu It is interesting how you connected the rotor and isolated it from the frame? In our case you can see in the image that we fixed the Pix to the solid structure. It is not possible for us to isolate or dampen the rotor because of two reasons. 1. It is class 50 machine which is pretty big. 2. We use internal combustion engine which needs a very sturdy mechanics hard connected to everything else. It is not an electric Helicopter. We are in preparation of a 2.5 meters 3 blade heli which will be twice as big and we build this class 50 one just as a proof of concept before we try the Pix on the larger machine. We are very very cautious because with the IC engine and the given size of the machines, we can make a lot of damage, if the machine goes loose in the air. I've seen people hurt by these blades... they are as knives with their kinetic energy.

rousselmanu, 2016-02-23:
Yes I see that your platform is a lot more dangerous than ours (which have only 35 mm of rotor diameter and is indeed electric...) which makes it a lot more difficult... In our case the rotors + the 2 motors + swashplate servomotors are on a separate platform which is linked with the rest of the structure using "silentblocs" like these ones:
[image]

Jomhuang, 2016-02-25:
@tubeme we just debug our mixer(CCPM) in Arco mode, and never try to switch Acro mode to manual mode.

tubeme, 2016-02-25:
Why not switching between Acro and Manual?

Jomhuang, 2016-02-26:
@tubeme if you want to switch between them, you should smooth the attitude setpoint for safety.

LorenzMeier, 2016-05-22:
Outdated.

closed 2016-05-22 by LorenzMeier