PX4/PX4-Autopilot #3789: MPU 6000- Data Ready implementation

Alexandreao, 2016-02-17:
Hello. 
I want to try to use the data ready interrupt from the mpu 6000 in order to read the data only when the sensor was new data.
For that i need to change the driver and set an interrupt event for the data ready pin of the mpu 6000
Can some one give me some tips how to do this?
Thanks

Inspirati, 2016-02-18:
i've just been working on adding magnetometer support to the mpu0250 driver
which is based on the mpu6000 driver.

my understanding is the interrupt is already enabled and being used, albeit
a little tricky to see just how.

for the int being enabled see mpu6000.cpp line 794:

// INT CFG => Interrupt on Data Ready

	write_checked_reg(MPUREG_INT_ENABLE, BIT_RAW_RDY_EN);        // INT:
Raw data ready

FAC94, 2016-02-18:
Hello @Inspirati, 

I'm working with @Alexandreao on some sensor drivers for the PX4 project and the purpose of having a pollrate defined by the data ready pin was to reduce the latency of the inertial data. 

From my understanding of the code, to use the data ready pin we need not only to configure the Data Ready generation on the sensor but also to define the reception of that signal by the software. Meaning that we need to initialize the sensor pollrate with the value SENSOR_POLLRATE_EXTERNAL on the ioctl function, but it seems to don't accept that value as you can see here: https://github.com/PX4/Firmware/blob/master/src/drivers/mpu6000/mpu6000.cpp#L1290-L1295.

Adding to that, the start() function is initializing an hrt_call with the value of the desired frequency of the sensor. That way is not very likely that we loose sensor data, but we have always a "fixed" latency with an unknown value, what is bad for data synchronization.

So @Alexandreao's question is about what should be done in order to the data ready interrupt generation be detected by the software.

Thanks

kd0aij, 2016-02-18:
have you quantified the latency which you wish to remove?

FAC94, 2016-02-18:
@kd0aij that's what I said on the previous answer, I don't know the latency because I don't have any way to know when the data was generated ( I think the sensor has no field for saving the data generation timestamp...), so measuring the data on a fixed time interval (assuming that is the same as the data generation) would give me a fixed latency, but not its value, as I don't know the difference between the timestamp of the data and the actual reading. 

If I had a data ready interrupt, I would know the time between the data generation and its reading...

LorenzMeier, 2016-02-18:
Because we sample at 1 KHz we know it's less than 1 ms. At what accuracy where you trying to synchronize?

kd0aij, 2016-02-18:
@FAC94 one thing I always do before filing an issue is to determine first the existence and then the magnitude of a problem. If the answer to either is "I don't know", then I do more research.

FAC94, 2016-02-18:
Hello @LorenzMeier,

The MPU6000 case is just an example to know if it is possible to implement the data ready. 

The accuracy we were trying to achieve was less than 1 ms (don't know if it is possible) that is not really a problem on the MPU6000 and on the gyroscopes and accelerometers in general (with an output rate greater than 1 kHz) but for a magnetometer, with an output rate of nearly 100 Hz, the problem scales from 1 ms to 10 ms, and than are the most critical cases where we wanted to try to reduce it.

FAC94, 2016-02-18:
@kd0aij 

The problem of "I don't know the value of the latency" is really the issue because if I knew that, I would just subtract it to the reading timestamp. I'm just trying to get some help on how to determine it, but I'll do more research and try to figure it out.

Thanks for your help.

davids5, 2016-02-18:
@FAC94 

I think it may be worthwhile to consider the system and the relationship of the clocking.

I think the mpu6000 has it's own clock and therefore is produces results that are asynchronous to the system clock. An interrupt solutions does not have 0 latency nor will it be fixed as the system data acquisition (sensor reading) is done off the HRT timer ISR and there is no nesting.    

Given the above and 1 ms sampling rate how does that relate the asynchronous sub-sensor (in the MPU6000) ready-rate?

It seams like dX/dT is what you should be looking at.

dagar, 2017-04-11:
Can we close this?

dagar, 2017-04-28:
Please reopen if there's more to discuss.

closed 2017-04-28 by dagar