PX4/PX4-Autopilot #3798: Track interference between RC and Wifi

LorenzMeier, 2016-02-18:
We have tested so far successfully FrSky and Robbe with parallel operation of the Wifi of Pixracer. We also have carefully tested Spektrum without issues. We still need to actively track the use of other RC systems in the 2.4 GHz band.

Pixracer has in-built HW to disable Wifi in flight. But we don't want to kill the usability for everybody without having actively seen issues.

careyer, 2016-02-18:
Hi Lorenz, I will try to get you the logfile of the crash-flight (it was not my copter but of a close friend). However I do not believe that this is an issue caused because of the WiFi interfering with FrSky.
The copter was parked in Loiter mode and immediatly fell out of the sky just the same moment when the QGC Android app opened the connection. It looked more like the Pixracer rebooted midflight because of the connection or something like this.
I will report back once I have the log. Cheers! - careyer

pkocmoud, 2016-02-18:
Careyer, was a power module used to power the PixRacer? If so which one?

careyer, 2016-02-18:
Yes, a power Module was used: ARSOVtech ACSP4

careyer, 2016-02-19:
Unfortunatly the MircoSD card was ejected from the cardslot of the Pixracer during the crash. The logfile is corrupted :-(

LorenzMeier, 2016-02-19:
@careyer Do you have the msgs_xxx file for that flight?

careyer, 2016-02-19:
@LorenzMeier Unfortunatly not. The MicroSD card was found lose within the body of the frame. It must have been ejected from the Pixracer druing the impact. Tried to read the card in a cardreader but it could not be read. :-( Broken filesystem. Had to do a reformat to get it working again. 

Taranis did not raise an "RSSI low" or "Telemetry Lost" alarm. Therefore we think it was not an RF interference. It looked more like the flightcontroler rebootet or reinitialized midair the moment the Android QGC APP tried to establish the WiFi connection. I am sorry that I cant provide more evidenace but only 3 broken Tiger CFK props and a bent motor shaft. :-|

DonLakeFlyer, 2016-02-19:
> The crash happend exactly in the moment that the QGC app was opened.

That would be the point where QGC starts to chew up bandwidth with communication over 2.4 out to the vehicle. Do you notice if parameter sync started? Did the toolbar change to show a connection? 

@LorenzMeier Now that I sort of have a simple file save/load solution on tablet (implemented for missions). I think it's time to turn on telemetry log saving for tablet. If we had that, we could see what was going on from the QGC side. I'll enter an issue.

dogmaphobic, 2016-02-20:
The ESP unit is constantly sending/receiving on the band. Yes, the data flow increases when the connection starts but if there was a radio interference issue, it would be noticed way before. Also, if you completely lost RC signal, it would go into RC Signal Loss mode. Not drop off the sky.

> The MicroSD card was found lose within the body of the frame. It must have been ejected from the Pixracer druing the impact. Tried to read the card in a cardreader but it could not be read. :-( (Broken filesystem). Had to do a reformat to get it operational again.

Too late now but on FAT/FAT32 file systems, it's fairly simple to rebuild the allocation table and restore the data. There are tons of tools out there to do that.

DonLakeFlyer, 2016-02-20:
> Yes, the data flow increases when the connection starts but if there was a radio interference issue, it would be noticed way before.

Not sure about that. If you are using a Spektrum DSM2 versus DSMX the increased traffic may hit the fixed 2.4 slots DSM2 is transmitting on. Wheras DSMX hops around, so you are less likely to have a problem. We've been having problems with a business that shoots a strongish 2.4 signal (all FCC legal) across the RC field I'm a board member on. Planes flying the lower end 2.4 systems (DSM2) fall out of the sky if you fly through the death zone at the time they are broadcasting. I've done tests with my own plane with multiple spektrum dsmx receivers with frame loss telemetry and it can handle the hits, but you can definetly see the frame misses. So my recommendation would be if you are running WiFi telemetry plus 2.4 radio then to stay away from older Spektrum DSM2 and stick with newer DSMX.

DonLakeFlyer, 2016-02-20:
And now you know the reason I've been so paranoid about the WiFi plus 2.4 radio thing. The board I'm on has been fighting with 2.4 interference issues at our field for a couple years now. We have our own spectrum analyzer continuous logging 24 hours a day to match the hits with the crashes.

hsteinhaus, 2016-02-21:
Hi Lorenz, I played a lot with WiFi on the copter, mainly to trigger cameras from onboard computers  (original goal was to stream live video to the ground). My observation was that the the radio link quality is sharply reduced even with the slightest amount of data transferred via WIFI. This is even true, if the radio channels to not overlap (e.g. with a modified WiFi firmware) and are in close proximity only. After many hours of testing, I came to the conclusion that is generally ok to trigger a camera with both tx and rx on the copter (short bursts of data), with tx power reduced to minimum on the onboard computer's wifi interface. Triggering from the ground already causes major radio glitches. I guess this is from the cameras response packets, as I could not find a way to throttle it's tx power. 

I tried both Futaba FASST (one way) as well as FrSky (two-way) radios. Futaba seems to be more robust against interference and has a better behavior of the failsafe flag (it seems to have some kind of rate limiting and some kind of good hysteresis), while the FrSky link seems to behave completely erratic sometimes. 

Streaming data (even telemetry) from the vehicle is a thing that I would definitely consider unsafe as soon as the vehicle is more than a few meters away from the pilot. Maybe it is ok for indoor flying. I moved back to 3DR radios for telemetry. After some frequency calibration (as described on diydrones), they perform very well here.

kd0aij, 2016-02-21:
@hsteinhaus What is the transmit power of the WiFi sources which cause problems for you? Are they FCC legal? Note that Phil's testing with 8266 indicated no RC problems at all (if I read his reports correctly).

hsteinhaus, 2016-02-21:
I have no experience with the 8266, I mostly used the Intel Edison module (embedded Wifi with PCB antenna), but also did some tests with TPLINK TL-WN722 USB sticks. Both are regular consumer grade WiFi devices @ 20dbm with CE compliance, I guess they are FCC compliant as well. 

BTW, the intensity of interference is mostly proportional to the average bandwidth usage. Varying the TX power has considerably less influence on the level of interference.

careyer, 2016-02-21:
@kd0aij : A FrSky Taranis was used, so FCC legal = YES ;-)

@LorenzMeier: We created a logfile today simulating a similar situation (this time the quad was straped to the workbench). Stabilze Mode at 20% Throttle. Then turned on WiFi and initiated a QGC connection via WiFi. This time nothing special happend other than it took very long to read all parameters from the quad (usually it takes 2-3sec.... this time 10-15sec).

My friend told me that when the crash happend the quad clearly was trying to keep stabilizing the aircraft but it seemed that it was otherwise pretty busy processing other stuff. He said that the quad drifted by huge amounts in means of angles of up to 15-20° until the stabilizing algorithm reacted and tried to keep it level again by giving reverse input. The aircraft was out of control for about 30-40sec without reacting on RC input and was going to fly away when it luckily hit a tree and fell to the ground.

pkocmoud, 2016-02-21:
I would expect that interference that killed the RC link would have triggered the receiver fail safe behavior, not an immediate crash?

v01d, 2016-02-22:
If I may add my two cents, I had experience in the past but with the opposite problem. I have a turnigy 9x receiver which uses FHSS (takes up the whole band) which immediately kills wifi. WiFi on the other hand does not bother RC since FHSS is quite robust. So the problem I had is that 2.4GHz wifi was unusable (and I had to resort to hacking the linux kernel to use my wifi adapter in the 2.3GHz range).

The opposite problem would probably happen with DSSS radios which mostly use a number of channels. If those channels happen to intersect with wifi, there will be a problem with RC I guess.

torb500, 2016-06-17:
Hi everyone, I experience a big interfereces issue between ESP and an AR8000 reciever from Spektrum : 
- With ESP powered on but PC not connected, everything works fine
- With ESP connected to the PC, I get a lot of "Manual Control lost" (every 3-4sec, which causes a throttle drop for about 200-500ms)
I absolutely need the wifi during the flight, and I saw somewhere that the ESP can be configured with a 5Ghz frequency, has anyone tried this solution or experience the same issue?

Thanks a lot, Nicolas

Buckler89, 2017-02-06:
> I saw somewhere that the ESP can be configured with a 5Ghz frequency, has anyone tried this solution or experience the same issue?

I'm afraid the the esp cannot work at 5 GHz [http://www.esp8266.com/viewtopic.php?f=6&t=4032](url).
See bottom of the page. :(

LorenzMeier, 2017-04-10:
It's mostly working fine, so closing this for now.

closed 2017-04-10 by LorenzMeier

IeiuniumLux, 2018-07-02:
I'm using PX4 v1.8.0 stack firmware with QGC v3.3.2, and I'm experiencing the behavior described by @torb500.

When the ESP module (firmware version 1.1.1) is not connected to QGC, then everything works fine.
But when QGC connects to the ESP module, then I get a lot of "Manual Control lost" and a latency ~200-500ms on the RC stick commands.

LorenzMeier, 2018-07-02:
Please make sure to create enough separation between RC and Wifi module or use an RC system on a different frequency.

v01d, 2018-07-02:
As a sidenote, I've been using dragonlink RC on 433Mhz for this reason exactly. an alternative I have not yet explored is to used WiFi on 5Ghz band (which would not work for ESP AFAIK).

IeiuniumLux, 2018-07-02:
@LorenzMeier By separation, do you mean between the RC (FrSky V8R4-II 2.4Ghz) receiver and the Wifi module?  If that separation is not sufficient enough, then what is the frequency recommended to use for the RC system?

asimonov, 2018-07-09:
we have same problem. FrSky/X8R RC pair loses signal the moment 2.4Ghz Wifi starts to transmit data in bulk.

IeiuniumLux, 2018-07-09:
It's definitely danger to fly with the WiFi module installed. During the vehicle setup or bench testing (SITL or HIL), then it's better to use the [BT module](http://ardupilot.org/copter/_images/Bluetooth.jpg). And if you need QGC to be connected to the vehicle during the flight, then it's safer to use the [SiK Telemetry Radio](https://store.mrobotics.io/product-p/mro-sikv2airgnd-mr.htm) instead.