PX4/PX4-Autopilot #3803: mocap/pose only updating position in local_position/pose .. orientation data wrong. Seems to be firmware problem.

shiohigari, 2016-02-19:
I am using an odroid with ubuntu 14.04 server running mavros to talk to my pixhawk running the px4 firmware. I can update the the pixhawk's local position correctly using the mocap topic however the orientation data is not updating and I find my drone cannot hold its yaw orientation with a slow drift over time. 

The mocap and local pos topics are shown below. 
I have the ATT_EXT_HDG_M parameter set to 2 and everything else going into the ekf set to 0 other than mocap. I tried setting ATT_EXT_HDG_M to 1 and using the vision topic instead but had the same result. 

Is this normal behavior?

[image]

wallarelvo, 2016-03-15:
Hi, I have also experienced this issue. It is because there is a feedback loop. You use the IMU to determine your orientation, then you do something to it (transform it or offset it), and you send it back to the IMU. The attitude q estimator then uses this to update the IMU, and so on. The attitude q estimator did not, however, update the /mavros/imu/data_raw or /mavros/imu/mag, so on your own computer, you can run imu_filter_madgwick and then there is no more drift. This at least worked for me.

chutsu, 2016-03-15:
Another thing to check, you might want to double check the Mocap data you are sending to Mavros. As the old very recently deprecated docs say (had to use Google cache to get that info back, link [here](https://webcache.googleusercontent.com/search?q=cache:Jv0i3KoisD4J:https://pixhawk.org/dev/ros/mavros_offboard+&cd=1&hl=en&ct=clnk&gl=ca&client=ubuntu)): 

> You need to send targets in ENU frame (ROS standard) which mavros internally converts to NED (PX4 standard) for sending to the flight controller.

chutsu, 2016-03-15:
Also what mocap system are you using?

closed 2017-04-02 by mhkabir