PX4/PX4-Autopilot #3808: VTOL transition altitude issues

sanderux, 2016-02-19:
@AndreasAntener @tumbili 
4 missions flown on ada's 20160218 tree. The behavior was consistent but there was still a big altitude drop in forward transition and a rise in altitude in back transition.
I was able to solve the wobble that occurred after back transition by lowering MPC_Z_VEL_MAX. It appears that the altitude rise after back transition, and the consequent error in altitude setpoint for the MC caused wobbling behavior when the max z velocity was set too high. 

Unfortunately the SD card failed halfway through the mission (only recorded a single identical byte). So i only have half a mission log: http://logs.uaventure.com/view/KiMU9DGHtyYpx3ipGMfNok

The mission was set to 25m, after transition it dropped +- 9m as shown below:
[image]

tubeme, 2016-02-19:
We observe the same loss in height during transition in 1.1.3.

AndreasAntener, 2016-02-20:
@sanderux It looks like you need to tune your FW pitch. It tries to command pitch up to catch the drop but the elevator it commands is only very little. Increase FW_PR_FF and FW_PR_P. Have a look at what I set in the 13005 config. Roll is better but could also be a bit tighter.

[image]

@tubeme could you try beta now? And if you want even better altitude hold set FW_T_SPDWEIGHT to 0 to give altitude priority over height. I got pretty good results with that.

sanderux, 2016-02-20:
@AndreasAntener Yes they were quite low, so that might be it. I'll test fly when weather permits. thanks

closed 2016-02-20 by sanderux