PX4/PX4-Autopilot #3810: Circle flight mode

gregswf, 2016-02-19:
One of the functions of PX4 controlled machine is to carry a camera it would be useful to have a circle flight mode activated with a switch.

lishidetiankong, 2016-02-20:
I agree with you, I think you should refer to APM Circle Mode!!!

liang-tang, 2016-03-09:
yes, and i made it.

liang-tang, 2016-03-09:
I had make it accroding to APM Circle Mode, it works well yet. and i added follow flight mode,but it works not good........

gregswf, 2016-03-09:
Isn't acceleration controlled already?
Would it work if there was an app calulating the gps cords on a circle of a given radius and a center point and feeding the cords as way points to the autopilot?
Then the auto pilot takes care of the speed acceleration and navigation to a waypoint?
Each waypoint would have a target heading calculated to point the circle centre.

gregswf, 2016-03-09:
I think the yaw rotation velocity would have to be calculated too.

liang-tang, 2016-03-10:
it's unsmooth to make a circle just via way points, but local position cords of circle can be calculated by mc_pos_control.

uavfans, 2016-03-10:
liang-tang  name is so familiar

szechyjs, 2016-03-22:
I would love to have the circle mode functionality of ardupilot.

http://copter.ardupilot.com/wiki/circle-mode/

liang-tang, 2016-03-23:
i had ported the code to px4 from ardupilot, and i would commit it in a few days.

gregswf, 2016-03-23:
@ Liang-tang, that's awesome, I can't wait to test it

DonLakeFlyer, 2016-03-23:
Make sure to let me know when this goes in so I can update QGC.

dagar, 2016-06-07:
Planned for v1.5.0 #3810

closed 2016-06-07 by dagar