PX4/PX4-Autopilot #3812: FW non RunwayTakeoff features

dagar, 2016-02-20:
@AndreasAntener @tumbili @LorenzMeier 

I was looking at adding a few features for fixed wing auto takeoff that I think will help with hand launching a larger plane and they mostly already exist within RunwayTakeoff. Specifically I'm interested in RWTO_HDG, RWTO_NAV_ALT, RWTO_MAX_PITCH, RWTO_MAX_ROLL. Reading the code I think RunwayTakeoff could be slightly reworked and be useful for all FW auto takeoff.

Any thoughts or objections before I take a stab at it? Does this interfere with any future work planned for RunwayTakeoff?

AndreasAntener, 2016-02-20:
@dagar so you would rework the runway takeoff class to the navigation helper for takeoff in general? I could see that work :). If the differences between 2 takeoff methods get too big you could also consider doing the same abstraction as for launch detection, but you would see if that makes sense or not on implementing I guess. On takeoff at least from my side is no work planned which would interfere with this.

One thing that I wanted to add back again now that the calculation for a waypoint with heading and distance from current location is correct [waypoint_from_heading_and_distance](https://github.com/PX4/Firmware/blob/master/src/lib/geo/geo.c#L318): for takeoff in current heading direction (RWTO_HDG 0), calculate a next waypoint some distance ahead and also use the navigator to give back yaw (would compensate for drift/crosswind) [getYaw](https://github.com/PX4/Firmware/blob/master/src/lib/runway_takeoff/RunwayTakeoff.cpp#L198). On the runway this is unlikely to happen but for the hand launch it could make a difference.

dagar, 2016-02-21:
I like your idea for RWTO_HDG 0. I'll come back to it once I've verified everything is working.

closed 2016-09-04 by dagar