PX4/PX4-Autopilot #3815: Altitude control misbehavior

owenqyzhang, 2016-02-21:
I am trying to use VICON for mocap position estimation following the instructions [here](http://dev.px4.io/advanced-fake-gps.html).

In the Step 6 from the tutorial, it says open "mocap_fake_gps.cpp", but there is no such file in the given location, so I guess it should be "mocap_pose_estimate.cpp". 

When I started testing after setting up everything, the drone dashes into the ceiling the moment I switch to ALTCTL mode, and the log is [here](http://logs.uaventure.com/view/rQmDGKA5qBVFRrbTkLAhAC). I noticed in ALTCTL mode, the LPOS.Z and LPSP.Z fits perfectly, so I guess it could be something wrong with the position control program, the altitude setpoint is constantly set to current position. The firmware I used is the v1.1.3 stable.

LorenzMeier, 2016-02-27:
@ChristophTobler Could you please have a look?

owenqyzhang, 2016-02-28:
A more recent test show the problem still exists in v1.2.0 stable. New log is [here](http://logs.uaventure.com/view/YnfbXk7z4Fg7x9UXfz3tvg). The position estimate seems fine, but altitude control is not working. When first switched to ALTCTL mode, the vel_sp is set to 0.5, but the drone is going upward, and the pos_sp is constantly reset to current position. When VICON is turned off, the altitude control works fine.

ChristophTobler, 2016-02-29:
@owenqyzhang Do you want to use mocap position estimation or do you want to fake gps? If you want to fake gps data you have to get the files from [here](https://github.com/mavlink/mavros/pull/442).
In the first log the LPOS.VZ has en offset. Was the version the only difference between the two? The second one looks much noisier. 
Can you maybe test if there is a delay from the MOCAP data? You could test it by sending MOCAP data but not using it for estimation. Then compare e.g. the estimated altitude with MOCAP.Z

ChristophTobler, 2016-02-29:
Not sure but maybe [this PR](https://github.com/PX4/Firmware/pull/3822) fixes the set point problem.

closed 2016-05-03 by tumbili