PX4/PX4-Autopilot #3818: Consider making RTL + Land default on RC loss?

LorenzMeier, 2016-02-21:
We should work out what's worse: RTL and Land or staying where it is. What do other systems use as default?

SimonWilks, 2016-02-21:
I've had this happen once. The receiver disconnected and it RTL'd. Luckily it was on a quad that doesn't fly too long and not the one that hovers for 40 minutes. After waiting between 5 and 10 Min the battery went out and it came down pretty quick. 

At one point while it was hovering above my reach it twitched a fair bit (probably a small gps glitch)  and I thought it was going to fly off. It didn't but I was pretty nervous having it up there knowing I had no control whatsoever.

Based on this experience my vote would be to RTL and land right away so as to reduce the amount of time that it spends in the air without the possibly of manual intervention.

gregswf, 2016-02-21:
If it stays hovering let's say above a forest ,then it will be difficult to get it home safely so defiantly RTL is a better option.

sanderux, 2016-02-21:
I think it depends on if it had a solid home lock before takeoff. and maybe a default waiting time of 1 minute before landing? this would give time to switch lipo's on the TX

tubeme, 2016-02-23:
I think @sanderux suggestion is the one, but I would put 2 minutes instead of one. I fly DJI professionally for video shoots and with DJI you never takeoff before you see 3 green lights ie a good GPS lock (with 9 or more sats locked). It is a kind of second nature to wait for this signal and not have red blinking led before takeoff. Also I've had numerous cases where I had RTL triggered and then during the RTL the machine comes closer to me and I get manual control of it. Also I've had a case with the RC bat failure and in the panic one minute was not enough for me to handle the situation. (as pilot I concentrate on the video and FPV flying not on failures, it takes some time to see problems and act given the adrenaline rush) It took me about 2-3 minutes to realize what the problem is, to go to my battery bag (in the car not at the home spot) to find the battery, to change it and then to turn on my RC.

jcooper21, 2016-03-26:
I would vote RTL and land for fixed wings based on wind direction.

julianoes, 2016-03-26:
I definitely vote for RTL + land for copters. It's one param that I have to change every time. I don't understand why you'd want it not to land.

For fixedwing, maybe loiter for a minute and then come in for land? Not decided there.

jcooper21, 2016-03-26:
The loiter for a minute should be part of the fixed wing land sequence to confirm wind direction at the home location so, it should circle over head before starting the landing decent.

LorenzMeier, 2016-04-21:
We have set this for now, closing.

closed 2016-04-21 by LorenzMeier

julianoes, 2016-04-21:
> We have set this for now, closing.

What does this mean? What's the action/consensus?

LorenzMeier, 2016-04-21:
It will land if you enable RTL on RC loss.