PX4/PX4-Autopilot #3824: Support rally points

sanderux, 2016-02-22:
Implement rally points (http://copter.ardupilot.com/wiki/common-rally-points/)
@DonLakeFlyer How are rally points transferred in missions in APM now?

DonLakeFlyer, 2016-02-22:
Rally points is a custom ArduPilot dialect mavlink message protocol. I think rally points need to be thought through with respect to landing sequences from DO_LAND_START. They seem to me to be kind of the same thing. It's odd that rally points are a special thing to me. Seems like they could be handled with landing sequences in a mission. But I guess you can send rally points without needing a mission.

sanderux, 2016-02-23:
I think failsafe landing points or "rally points" as APM calls them is an important feature to have during autonomous flight. A fixed wing should be able to land safely (or at least make an attempt) when thrust is lost or lipo is critical. VTOL should also be considered as this might involve a failsafe transition to MC, the MC might not be able to reach the home point. Failsafe landing zones should indeed not be limited to missions as RTL mode could be engaged from several modes. 

For FW  we could consider a few approaches;
1. add a separate DO_LAND_START for every failsafe point
2. Have the failsafe landing point contain a preferred approach heading
3. Land in a spiral and assume enough clearing around the failsafe point
4. A combination of the above

@DonLakeFlyer Are rally points in APM stored separate from missions? and does QGC support control over them separate from missions? 

@LorenzMeier @tumbili any thoughts?

DonLakeFlyer, 2016-02-23:
QGC does not currently support rally points. Rally points are unrelated to missions in APM as far as I can tell.

dagar, 2016-03-15:
I've heard a few requests for rally points and their use in fixed wing landing.
I believe it would be something like for landing fly to the rally point, circle down, and break out where appropriate on the glide slope for landing approach.

I think the other use is as a home substitute for RTL?

If we can work through the details of rally points I could implement the fixed wing landing piece.

DonLakeFlyer, 2016-03-15:
I believe rally points in ardupilot are a multi rotor thing. Fixed wing is DO_LAND_START.

dagar, 2016-03-15:
hmm, it sounds like we need MAV_CMD_DO_LAND_START implemented in PX4.

dagar, 2016-07-06:
@sanderux I've started implementing MAV_CMD_DO_LAND_START
Do you want to revisit this and discuss exactly what you need?

So far it's going to be
 - as a mission item MAV_CMD_DO_LAND_START marks the start of the landing sequence in a mission (approach + land)
 - as a command it initiates AUTO at MAV_CMD_DO_LAND_START. I'm also considering making the AUTO_LAND mode do the same
 - then we can combine RTL (failsafes) with landing properly through the predefined approach + land

I haven't through how supplying LAT/LON with sending a MAV_CMD_DO_LAND_START command should work.
Do you think we need multiple failsafe points?

sanderux, 2016-07-06:
@dagar For commercial applications failsafe sites will be a must-have.
According to the mavlink spec the MAV_CMD_DO_LAND_START initiates a landing at the closest landing site. When given coordinates it would be the closest to that location. but i think there is some nuance in this, first let me explain what my idea on failsafe sites (or rally points) is: 

I think rally points is a good start but rally area's would be better. You could think about the following scenario: A plane performs coastline inspections, for visibility and safety reasons it will fly over water. When the FC detects a problem with thrust it can start the failsafe procedure and land. It should not land in the water for obvious reasons. Depending on the location either the beach or the terrain just behind the beach would be a good place to land, no specific location but an entire area. The same would be true for road, electricity, rail or oil pipeline inspections.

On the nuance of choosing a landing site i think we should consider the both feasibility and the reason for initiating the landing. Obviously the designated landing site would be preferred over a failsafe site, even if a failsafe area might be closer. You could judge that based on why the land_start was initiated (was it actually a failsafe event) or decide based on feasibility (or both),

As i'm considering all these scenario's i'm not even sure if all of this should be managed by the FC, maybe a sidekick computer should take care of this. Let me know what you think on this matter.

dagar, 2016-07-06:
I'll try to stay on topic for this particular issue and focus on rally points. There are a lot of different areas I want to improve for fixed wind autonomy and failsafes, but right now let's just try to determine if we want APM style rally points or not.

So if I'm flying a fully autonomous mission and hit a failsafe where the plane is still functional (loss of comms, low battery threshold, etc) I think what I almost always want is to fly home and land safely at the preplanned approach + landing waypoint.

Now if that's not possible and I can't make it home (bad motor, GPS, etc) what should happen? Would I go through the planned mission and periodically insert rally points for safe landing spots? Or would it make more sense to extend the geofence system so that's it's factored in and respected during the failsafe landing modes (AUTO_LANDENGFAIL, AUTO_LANDGPSFAIL, DESCEND, etc). Assuming you still have gps...

For now (over the next week) I'm going to implement MAV_CMD_DO_LAND_START and MAV_CMD_DO_GO_AROUND, which gives me safe auto lands in the event of loss of comms, etc. Let's try to think through what we'd like to have eventually.

Right now I'm not sure if I'd ever use APM rally points. What do you think?

LorenzMeier, 2017-04-10:
Geofence made it in but it still WIP, moving this.

lfelipe, 2017-07-18:
@LorenzMeier was this integrated or need to have the milestone changed?

LorenzMeier, 2017-07-18:
@lfelipe Geofence is in master, but the rally points (while supported on the protocol level) need someone to push for regarding testing. It would not be hard and can be completely wrapped up in SITL.

LorenzMeier, 2017-07-18:
Current discussion is here: https://github.com/PX4/Firmware/issues/7565

closed 2017-07-18 by LorenzMeier