PX4/PX4-Autopilot #3825: Investigate quadrotor prop failure recovery

sanderux, 2016-02-22:
Investigate the abillity to do a controled landing after prop failure on quadcopter. http://youtu.be/ek0FrCaogcs

sanderux, 2016-02-23:
@tumbili how can we identify loss of a prop or motor failure?

mhkabir, 2016-02-23:
@sanderux I'd be interested in this too. Feel free to pull me in when you have something going. 

As for detecting motor failure, ESC feedback would be optimal. UAVCAN ESCs should be able to do this with ease. @pavel-kirienko

sanderux, 2016-02-23:
@mhkabir I think the sudden loss of inertia would be enough indication. I also think that simply letting go of heading hold (let it spin the way it wants) might even be enough to allow a controlled descent.

tubeme, 2016-02-23:
@sanderux I watched this demo when it was released 2 years ago. DJI use similar algorithm for their A2 but it is less redundant if I can put it this way. Their algorithm can sustain 1 prop failure in hex configuration and up to 2 props in octo mode. Also Zero Tech is using similar algorithm as DJI in their Gemini redundant controller. @mhkabir I don't think that the ESC feedback is a good idea in this case because here we are talking about aggregate failure not motor failure ie. we can have a motor failure, ESC falure and broken prop all these are failures that should trigger the algorithm. Also you narrow down the application only to UAVCAN ESC's, and now this technology is just in its infancy and there are almost no users of UAVCAN esc's for now.

mhkabir, 2016-02-23:
@sanderux @tubeme Good points. I agree with you, but I'm sure that feedback from ESCs opens up  multiple new avenues!

Slightly OT - 
https://www.youtube.com/watch?v=pGU1s6Y55JI
and related, @priseborough's work on APM : https://www.youtube.com/watch?v=4e9KAhY01V4

Would anyone be interested in collaborating on a similar project for PX4? I'm about to dive into that implementation from tomorrow, and having a partner to work with would be great! It would be simpler, and initially use GPS. The idea would be to enable throw-launches of small quadrotors, with an aggressive recovery and position hold.

sanderux, 2016-07-23:
@mhkabir i wont be working on this atm. please open a new issue if you want to pursue the hand launch

closed 2016-07-23 by sanderux