PX4/PX4-Autopilot #3834: VTOL: use filtered airspeed for transition condition

AndreasAntener, 2016-02-23:
We transition too fast in gusty conditions, increasing the risk of stalls.

sanderux, 2016-02-23:
Should consider this before changing https://github.com/PX4/Firmware/commit/aa90c0686b766102b3bd1efb2a4f7ecd28adbe61

LorenzMeier, 2016-02-24:
@sanderux No, that's entirely unrelated. You need to have TECS and the VTOL control to use the same quantities, because both use airspeed to ensure sufficient lift. They can not be using different ones, only TECS can *also* use true airspeed to ensure a *mission profile* which included traveling fast enough, but that has nothing to do with having enough lift.

sanderux, 2016-02-24:
The commit i referrenced states exactly what you say. Thats why i referenced it ;)

AndreasAntener, 2016-02-24:
I'd like to add, after another round of log analysis, that the 2 flights where our plane stalled during the transition had nothing to do with a gust induced airspeed spike but in-fact only with the (now fixed) TECS pitch integrator bug.

Using the filtered airspeed still makes sense though. We could think about taking the filter from TECS and feed it into the control state somewhere.

tubeme, 2016-02-24:
Here is a Log from the same issue. Early stop of quad motors due to high wind creating stall. 

http://logs.uaventure.com/view/3vg29z4WwQYkfEb5gLq9ak

What is the point of setting front transition time when the switch is made by the airspeed? I don't know how it is coded. If it is set that either one of these conditions trigger the quad stop. We think it is better to make the transition such that it will be at least the time of the front transition time and then after that to check if the airspeed is achieved. This way we guarantee at least # of seconds preset for the transition independent from the weather conditions. Because otherwise just ONE gust of wind could brake camera for 15K US :).

LorenzMeier, 2016-02-24:
You have set the transition speed to stall speed: ```VT_ARSP_TRANS = 10.0``` that's a very fundamental issue. You should set the transition speed to more than your FW TRIM speed.

I also think that for your particular airframe the speeds are very low and likely very close to stall, so the slightest wind gust will stall your plane.

AndreasAntener, 2016-02-24:
@tubeme 
- I agree with Lorenz on transition speed
- Could you update to beta or at least v.1.2 for your tests? We just tested v1.2 and we get very good performance if everything is tuned and properly setup.
- In this log you shared, is looks like it was already destabilized just before the transition happened, and thrust was reduced almost completely.

Also tuning: I looks like it was windy, but your system didn't seem to try very hard to hold the attitude setpoint. That is why it's wobbling around.

[image]

Here's an example in similar conditions with our Fun Cub. Note that the quad setup on our Fun Cub is not well sized (too weak). You can see that all motors are close to saturation but it still holds attitude in almost 7m/s wind. (until it lost it because of complete thrust saturation when the conditions exceeded the systems limits eventually)

[image]

In general: from the logs I saw so far, the airspeed filter would need to have a pretty large time constant to actually make a difference. Apart from Sander's malfunctioning airpseed sensor, there were never really  "spikes" as in a jump of 3-5m/s in below 1 second. So we need to find out what a gust looks like to determine the filter properties. Also what TECS does internally looks like it would not be enough.

tubeme, 2016-02-24:
You are very correct! We are making endurance testing so we want the least amount of quad during the transition but still on the safe side. Our Stall speed is about 7-8 m/s and we are comfortably flying it in manual at that speed 7-10m/s. For the testing we gave a little more in the parameters ie. FW_AIRSPD_MIN = 10.0. For the transition in this conditions we need 4 seconds never had any problems with the KK2 just 4 seconds made with script in the Taranis to transfer throttle. And this way we have 1:30h of run with Pixhawk in Manual. So far in Loiter we make 1:10h. Of course there is an extra cost in Loiter because of servo work and the plane banked all the time which reduces the efficiency. We preset the throttle during trnasition so we know the trust for this 4 seconds. I expect if after this 4 seconds at my preset throttle then if the airspeed is not achieved the PX4 to apply stall prevention technique by dipping a little and applying 100% throttle.

AndreasAntener, 2016-02-24:
@tubeme Oh btw on your other question, the front transition timeout is only used to ramp up fixed-wing throttle. It might actually not be such a bad idea to wait with the transition until that finished, still thinking about the consequences though when there actually is steady wind.

tubeme, 2016-02-24:
@LorenzMeier @AndreasAntener Sometimes I think the simpler the solution the better. Why do I preset the VT transition throttle and VT time for front and back transitions if they are not constants? The the transition procedure is a constant! with one exception the airspeed. Detecting WIND and its direction is a very very big and interesting topic but it is pretty complex and should be dealt in the future when there is time for this I think. There are Autopilots that can detect wind and calculate at what spot to launch the parachute in order to reach a specific landing spot. For the transition I think just 4 seconds preset throttle is perfect, then after we see the condition for air speed. If the craft is at higher altitude we apply the stall recovery algorithm if the craft is at lower altitude we recover with quad. BTW I think that there is a need for more thorough work on stall recovery algorithms! Different techniques for different frames!

tubeme, 2016-02-24:
The steady wind does not mater. If the wind is front one I will just be at trim speed after the transition, no problem with that. If back wind I might experience stall approach or stall in very high gusts. Both conditions need stall recovery. I guarantee you if the quad is holding the plane for 4 seconds I will generate a lot of speed even in strong weather. You see our throttle is at 0.7 because of economy and also because we want it smooth. Our transition looks like Star Wars transitions I swear :)

AndreasAntener, 2016-02-24:
We're not talking about wind detection or anything complex here, rather about the simple fact that you need airspeed to fly.

Let me ask you the following question.

Would you rather want to sell a product that transitions like this:
https://youtu.be/fqSvGcRu0YY?t=67
(sorry, couldn't find a better video)

Or like this:
https://youtu.be/eMhoDWZn5TM?t=45

tubeme, 2016-02-24:
We are perfect with just the seconds. Or at least can you add a parameter that we can preset the time that the quad should hold during the transition? With this for our case we will not need gust detection. Just this parameter...

tubeme, 2016-02-24:
Of course it is the second Video that I want to sell :)

AndreasAntener, 2016-02-24:
:)

You know, I'm not changing my mind about needing airspeed here ;) but considering the following assumptions:
- a smallish airframe that is able to accelerate to cruise speed in a few seconds
- and that you fly in completely manual control

Then I could argue for your case in that we currently have a conceptual break for the transition because you need airspeed but for the rest of the flight you don't. So maybe this is a valid option in your case. But as soon as we're talking about more than just attitude stabilization, airspeed is necessary.

LorenzMeier, 2016-02-25:
@tubeme You used incorrect settings for the controller. The equivalent of your throttle ramp is just setting a higher transition speed.

If you have tail wind the approach we have implemented will not transition if you don't reach enough speed to safely transition.

tubeme, 2016-02-25:
OK we will change the VT_ARSP_TRANS to higher value. We will report back. Wind is problem though we've had similar cases with VT_ARSP_TRANS =12 which is our TRIM speed. We have to remember that until we reach steady stream of air at 1-2 m/s over the stall speed we have no lift in the wings at all. And if a gust triggers the switch the plane becomes uncontrollable. Given that we  make the transition at 5-10 meters in the air it is certain crash. And we want it low because we want to save energy.

sanderux, 2016-02-25:
We could choose to support both considering we will probably implement minimum transition time (https://github.com/PX4/Firmware/pull/3763#issuecomment-188514226) , that combined with an (explicit or implicit) transition airspeed of 0 would give you the result @tubeme suggests. 

it would indeed not be safe in a tail wind condition or if the timeout is too low.

sanderux, 2016-02-25:
@tubeme the minimum transition time will also address the wind gust situation

sanderux, 2016-02-25:
I will have a PR tomorrow: https://github.com/PX4/Firmware/issues/3850

LorenzMeier, 2016-02-25:
@tubeme Start with setting your transition speed with something higher than trim and just below max. I'm pretty sure your throttle ramp overshot 12 m/s by far, so you might as well shoot to achieve the same.

sanderux, 2016-02-25:
@tubeme set your MPC_Z_VEL_MAX to 1.5, this will fix your backtransition wobble problem

tubeme, 2016-02-25:
@LorenzMeier @sanderux Thank you! We will put 14 m/s and change the parameter.

sanderux, 2016-02-25:
and i would increase your VT_B_TRANS_DUR to 3.5 for an even smoother back transition

tubeme, 2016-02-25:
You might laugh but we want to mimic what the birds are doing. Their transition starts almost immediately.

LorenzMeier, 2016-02-25:
@tubeme Don't forget that your attitude tuning also looks bad. You should have better attitude control in hover.

tubeme, 2016-02-25:
@LorenzMeier We have more work on the PID tuning for the quad that is for sure. BTW my partner is coming to Bulgaria so we will concentrate on PID tuning of the Quad for the following couple of days. Lorenz I still cannot calibrate my magnetometer...

https://github.com/PX4/Firmware/issues/3711

I tried everything and it gives the same excessive offset error. Tried disabling the external one and calibrating only the internal one nothing changed... Really I'm clueless...

sanderux, 2016-03-02:
@tubeme do you have any more logs on recent versions of the firmware?

dagar, 2016-11-13:
What's the current state of this? Should we centralize the filtered airspeed and publish from sensors? Could we just use EKF2 and control state airspeed everywhere?

LorenzMeier, 2017-01-04:
@dagar @AndreasAntener Can we close this?

AndreasAntener, 2017-01-05:
It's still a topic. The very cheap solution for VTOL is we just do a very slow lowpass over airspeed to make sure we get some more safety. But it's not the best solution. I could imagine having some form of transition progress state estimation in EKF taking airspeed and velocities into consideration so we don't just rely on one sensor. This would benefit the airspeed-less operation as well.

tubeme, 2017-01-08:
@LorenzMeier @AndreasAntener @sanderux @dagar 

So far we have a very good results with 1.5.0. Indeed such a good results that yesterday we've had a hardware problem with one of the motors (the mount just turned 30 degrees in flight because it was not tight enough) and the vtol just span on hover at one spot (but controlled spin). So we just wanted to see if it could make a transition. And it did it just perfect transition which looks very stable even with bad hardware problem.

But there are some preconditions. 1. Perfect PID tunning.  2. Perfect Airspeed calibration.

We still have huge problem calibrating the airspeed in wind condition on the field. So what we did is make a table with the relation between temperature and pressure and change the airspeed calibration manually with the parameter on the field. Thus far the system works but it is not novice proof, and it could not be for a users.

We could not calibrate the airspeed with 1.5.2 and 1.5.3 even on a bench test. Tried probably 30 times So without airspeed sensor both 5.2 and 5.3 perform terribly given the VTL3 we are working on is 2.1m wingspan.

dagar, 2017-04-28:
Closing this particular issue. I'm thinking of adding additional transition conditions. Later we should find a way to sync this with the airspeed TECS uses.

closed 2017-04-28 by dagar